fpoussin/QtUsb #4: SetupDiEnumDeviceInfo  bad api

linminglu, 2016-10-22:
SetupDiEnumDeviceInfo Can not get the deivice on usb type

fpoussin, 2016-10-24:
Can you be more specific?
Any test case?

Thanks

kangqf, 2016-11-24:
right i get the same problem,the  SetupDiEnumDeviceInfo always return "ERROR_NO_MORE_ITEMS 259 (0x103) No more data is available."  for all devices

kangqf, 2016-11-24:
oh i'm so sorry, i use the ClassGuid but DeviceGUID,so it couldn't open the devices.

linminglu, 2016-11-28:
In this function   the  winusb_initializen . return 0 !!!
    PWINUSB_INTERFACE_HANDLE tempHandlePtr = phWinUSBHandle;
    if(!WinUsb_Initialize(hDeviceHandle, tempHandlePtr))
    {
        printUsbError("WinUsb_Initialize");

linminglu, 2016-11-28:
and getlasterror is 0

fpoussin, 2016-12-02:
Check the function doc here:
https://msdn.microsoft.com/fr-fr/library/windows/hardware/ff540277(v=vs.85).aspx

Are you sure the interface is not already claimed by a driver or application?

fpoussin, 2017-02-13:
No updates - closing

closed 2017-02-13 by fpoussin