fpoussin/QtUsb #5: Platform not supported

slavkoja, 2016-12-03:
I tried the git and latest version, in both i get:
```
qmake 
make
cd src/ && /usr/lib64/qt4/bin/qmake /home/XYZ/tmp/QtUsb-0.2.0/src/src.pro -o Makefile
Project ERROR: Platform not supported!
Makefile:33: návod pro cíl „src/Makefile“ selhal
make: *** [src/Makefile] Chyba 2
```
I am on latest Funtoo Linux with gcc 5.3, please what i can do with this?

fpoussin, 2016-12-05:
Seems like this is a bug in qmake. Can you verify _/usr/lib64/qt4/bin/qmake_ is present?

slavkoja, 2016-12-05:
of course, i can:
```
file /usr/lib64/qt4/bin/qmake
/usr/lib64/qt4/bin/qmake: ELF 64-bit LSB executable, x86-64, version 1 (GNU/Linux), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, stripped
```
```
/usr/lib64/qt4/bin/qmake --version
QMake version 2.01a
Using Qt version 4.8.7 in /usr/lib64/qt4
```
Output of ``qmake -d`` attached
[qtusb.txt](https://github.com/fpoussin/QtUsb/files/631328/qtusb.txt)

fpoussin, 2016-12-05:
I pushed a fix and release 0.2.1, that should work now

slavkoja, 2016-12-05:
Thanks, seems to works, make was run without errors. :)

closed 2016-12-05 by slavkoja