fpoussin/QtUsb #6: getAvailableDevices() returns empty list.

tswindell, 2017-02-13:
I'm presuming I'm doing something wrong, but why do I get no results from this:

    #include <QCoreApplication>
    #include <QUsb>

    int main(int argc, char *argv[])
    {
        QCoreApplication a(argc, argv);
        QUsbManager devices;

        QtUsb::FilterList list = devices.getPresentDevices();
        foreach(QtUsb::DeviceFilter filter, list)
        {
            qDebug("%04x:%04x", filter.vid, filter.pid);
        }

        return a.exec();
    }

fpoussin, 2017-02-13:
Which platform are you on?

tswindell, 2017-02-13:
Windows 10

fpoussin, 2017-02-13:
Are you sure you have unclaimed WinUSB devices available?

tswindell, 2017-02-14:
Well, the device shows up when it's plugged in through the deviceInserted signal. It's a HID device, which should be unrecognised by the system. But I guess HID devices can be automatically grabbed by Windows?

tswindell, 2017-02-14:
It's actually a composite device, it exposes some HID interfaces which seem to be recognised by the HID subsystem in Windows. Though it also has 3 unrecognised devices.

fpoussin, 2017-02-14:
I think I found the issue, I'll let you know.

tswindell, 2017-02-14:
Ok, so I just noticed QUsbDevice::getAvailableDevices(), which returns what I need. It seems that QUsbManager is actually a QThread so maybe it isn't populated immediately like my example expected.

fpoussin, 2017-02-14:
Yes, getPresentDevices() does not populates from the start, that's the problem.

fpoussin, 2017-02-14:
So the documentation is a bit misleading, getPresentDevices() actually returns devices that are present in the system AND in the list. You have to use addDevice() to add these to the list.
It's meant to only return devices that you are interested in.

tswindell, 2017-02-14:
Okay, that's fine. :)

fpoussin, 2017-02-14:
Feel free to close the ticket.
I'll update the documentation.

closed 2017-02-17 by fpoussin