fpoussin/QtUsb #25: Why do HTML documents generated with qdoc turn QString into int？

xinchangli, 2019-04-01:
//.h
[image]

//cpp
[image]

//.qdocconf
[image]

html file
[image]

fpoussin, 2019-04-01:
Hi,
I'm not sure how this is related to QtUsb, could you explain?

Thanks

xinchangli, 2019-04-02:
sorry,wrong place @fpoussin

closed 2019-04-02 by xinchangli