fpoussin/QtUsb #28: Packet loss

zhangzilong051524, 2019-06-04:
Hello, have you ever experienced the problem that the packet received is not covered by the next packet in time, resulting in packet loss

fpoussin, 2019-06-04:
Hi,

Never had this happen, which version are you using?

zhangzilong051524, 2019-06-06:
> Hi,
> 
> Never had this happen, which version are you using?

I used the 1.0.22 version. I used a different thread to call "libusb_handle_events_timeout_completed", and then used the receive callback to receive the data.
Receiving code:

void QUniversalSerialBus::AsyncRead()
{
    int rc = 0;
    m_read_transfer_mutex.lock();
    qTransfer = libusb_alloc_transfer(0);
    readBuffer.resize(64*2);
    libusb_fill_bulk_transfer(qTransfer,dHand,130,reinterpret_cast<uchar *>(readBuffer.data()),64*2,&callbackRevc,this,0);

    rc = libusb_submit_transfer(qTransfer);
    if(rc < 0)
    {
        libusb_free_transfer(qTransfer);
        qTransfer = nullptr;
        qDebug() << "libusb_submit_transfer file:" << rc;
        return;
    }
}

void QUniversalSerialBus::callbackRevc(struct libusb_transfer *transfer)
{
    QByteArray s((char*)transfer->buffer,transfer->actual_length);
    QString str = s.toHex().data();
    qDebug() << str;
    QUniversalSerialBus* qObj = static_cast<QUniversalSerialBus*>(transfer->user_data);
    if(transfer->status == LIBUSB_TRANSFER_COMPLETED)
    {
        qObj->Reviceiver(transfer->buffer,transfer->actual_length);
    }

    libusb_free_transfer(transfer);
    transfer = nullptr;
    qObj->qTransfer = nullptr;
    qObj->readBuffer.clear();
    qObj->m_read_transfer_mutex.unlock();
    qObj->AsyncRead();
    return;
}

void QEventHandler::run()
{
    timeval t = { 0, 0 };
    while(!stop)
    {
		//context ：Call "libusb_init"
        if(libusb_handle_events_timeout_completed(context, &t, Q_NULLPTR) < 0)
         {
              qDebug() << "libusb_handle_events_timeout";
         }
    }
}

fpoussin, 2019-06-06:
This is not QtUsb, you need to look for libusb support.

closed 2019-06-06 by fpoussin