fpoussin/QtUsb #29: Windows how to install Qt Usb module?

neomissing, 2019-06-21:
I have downloaded the file（qt5.12-usb-msvc-x86、qt5.12-usb-msvc-x64、qtusb-0.5.0-beta2.qch）, but I don't know how to install it on Windows. Is there any specific instructions? Or download the decompression overlay directly into QT?

fpoussin, 2019-06-23:
You have to decompress the archive where your Qt folder is. (not inside)
In my case it's C:\ since my Qt install is in C:\Qt

neomissing, 2019-06-24:
> You have to decompress the archive where your Qt folder is. (not inside)
> In my case it's C:\ since my Qt install is in C:\Qt
> You can put the qch file wherever you like, you have to add it to Qt creator manually anyway. (in the settings)

Thank you, I have already installed according to the instructions, but the following error occurred in 5.13. I would like to ask if you can provide the latest support for version 5.13.
[image]

fpoussin, 2019-06-24:
Well if you try code compiled for Qt5.12 it's not going to work with Qt5.13 so this is normal.
I will update the release with Qt5.13 binaries.

neomissing, 2019-06-25:
> Well if you try code compiled for Qt5.12 it's not going to work with Qt5.13 so this is normal.
> I will update the release with Qt5.13 binaries when it's available on AppVeyor.

Thank you very much, look forward to updating！

neomissing, 2019-07-06:
Hello, I want to ask when it can be updated, thank you！

neomissing@163.com
 
From: Fabien Poussin
Date: 2019-06-25 01:36
To: fpoussin/QtUsb
CC: neomissing; Author
Subject: Re: [fpoussin/QtUsb] Windows how to install Qt Usb module? (#29)
Well if you try code compiled for Qt5.12 it's not going to work with Qt5.13 so this is normal.
I will update the release with Qt5.13 binaries.

fpoussin, 2019-07-06:
As I said, it depends on AppVeyor. By the look of it it's going to take at least a month.
You can build the binaries yourself if it takes too long. See README.md

fpoussin, 2019-08-02:
I've release binaries for 5.13
https://github.com/fpoussin/QtUsb/releases/tag/0.5.0-beta3

closed 2019-08-02 by fpoussin

neomissing, 2019-08-03:
thanks.

neomissing@163.com
 
From: Fabien Poussin
Date: 2019-08-03 06:57
To: fpoussin/QtUsb
CC: neomissing; Author
Subject: Re: [fpoussin/QtUsb] Windows how to install Qt Usb module? (#29)
I've release binaries for 5.13
https://github.com/fpoussin/QtUsb/releases/tag/0.5.0-beta3

neomissing, 2019-08-06:
Hello, may I ask if QtUsb supports hid? My test writing can't work? If so, can I give a demo? Thank you！

neomissing@163.com
 
From: Fabien Poussin
Date: 2019-08-03 06:57
To: fpoussin/QtUsb
CC: neomissing; Author
Subject: Re: [fpoussin/QtUsb] Windows how to install Qt Usb module? (#29)
I've release binaries for 5.13
https://github.com/fpoussin/QtUsb/releases/tag/0.5.0-beta3

fpoussin, 2019-08-07:
Hi,
HID is something completely separate from USB and is not supported by QtUsb.

neomissing, 2019-08-07:
Thank you very much for telling me, then I am using hidapi！

neomissing@163.com
 
From: Fabien Poussin
Date: 2019-08-07 18:48
To: fpoussin/QtUsb
CC: neomissing; Author
Subject: Re: [fpoussin/QtUsb] Windows how to install Qt Usb module? (#29)
Hi,
HID is something completely separate from USB and is not supported by QtUsb.