fpoussin/QtUsb #30: Android build

gaminn, 2019-08-01:
Hi,
will QtUsb work on Android to read generic HID USB device?

Are there any instruction how to build this library for crosscompiling on Windows host for Android armv7 device, Qt 5.11.2?

fpoussin, 2019-08-01:
Hi,

I've never tested it on Android.
You should be able to cross compile it using the Qt SDK for android.

tranvinhphuc111996, 2019-08-02:
Hi, i need this lib for android. i cant build this module for android_armv7

fpoussin, 2019-08-02:
I can't help you if you don't give me any information.

fpoussin, 2019-08-06:
I've added experimental support, it now builds and can be used like other platforms.
You will need to initialize the libusb submodule.
Untested.

tranvinhphuc111996, 2019-08-07:
/home/phuc/Desktop/UsbHid/QtUsb/QtUsb/src/usb/qusbinfo_p.h:9:31: fatal error: libusb-1.0/libusb.h: No such file or directory
 #include <libusb-1.0/libusb.h>
!!! Hi !!! i found this error when i make build

tranvinhphuc111996, 2019-08-07:
Oh i was built .

closed 2019-08-07 by fpoussin

HKH13, 2021-12-14:
Hi, can i guide me on how can I install qusb?
because I can not install it on linux as guide you wrote