fpoussin/QtUsb #34: How to use on MacOS?

jhihn, 2019-09-26:
I was able to compile it by adding:

    export CPLUS_INCLUDE_PATH=/usr/local/Cellar/libusb/1.0.23/include

Which is the location where Homebrew is installing libusb-1.0. 

Then running the notifier example, inserting a iPhone 8 it says:

    Starting...
    Press CTRL+C to close.
    hotplugCallback
    Could not open new USB device: -4

This however works for s USB/Serial adapter that I have:

    Starting...
    Press CTRL+C to close.
    hotplugCallback
    devices inserted
    V0403:P6001

But libusb can see it:
    
    ***[ void QUsbInfo::checkDevices() ]***
    [ 6.056919] [00000307] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
    [ 6.056930] [00000307] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
    [ 6.056936] [00000307] libusb: debug [handle_events] poll() 1 fds with timeout in 0ms
    [ 6.056951] [00000307] libusb: debug [handle_events] poll() returned 0
    [ 6.091505] [00002107] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xaf9ad4d868ca
    [ 6.091887] [00002107] libusb: debug [darwin_get_cached_device] parent sessionID: 0x12929ff34e64
    [ 6.091896] [00002107] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xaf9ad4d868ca/0x14623100 against cached device with sessionID/locationID 0x1292d67c542e/0x14623200
    [ 6.091900] [00002107] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xaf9ad4d868ca/0x14623100 against cached device with sessionID/locationID 0x1292acdb4fac/0x14624100
    [ 6.091904] [00002107] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xaf9ad4d868ca/0x14623100 against cached device with sessionID/locationID 0x12929ff34e64/0x14623000
    [ 6.091907] [00002107] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xaf9ad4d868ca/0x14623100 against cached device with sessionID/locationID 0x12929cdc224c/0x14624000
    [ 6.091910] [00002107] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xaf9ad4d868ca/0x14623100 against cached device with sessionID/locationID 0x129299c751d9/0x14622000
    [ 6.091914] [00002107] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xaf9ad4d868ca/0x14623100 against cached device with sessionID/locationID 0x129296ac88f6/0x14621000
    [ 6.091917] [00002107] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xaf9ad4d868ca/0x14623100 against cached device with sessionID/locationID 0x12925cd603e2/0x14620000
    [ 6.091920] [00002107] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xaf9ad4d868ca/0x14623100 against cached device with sessionID/locationID 0xab7b75a8/0x14613000
    [ 6.091924] [00002107] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xaf9ad4d868ca/0x14623100 against cached device with sessionID/locationID 0x9c3ad1ad/0x14640000
    [ 6.091927] [00002107] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xaf9ad4d868ca/0x14623100 against cached device with sessionID/locationID 0x9aee0842/0x1212000
    [ 6.091931] [00002107] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xaf9ad4d868ca/0x14623100 against cached device with sessionID/locationID 0x89ebe5d6/0x14610000
    [ 6.091934] [00002107] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xaf9ad4d868ca/0x14623100 against cached device with sessionID/locationID 0x88eac505/0x1211000
    [ 6.091937] [00002107] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xaf9ad4d868ca/0x14623100 against cached device with sessionID/locationID 0x7505644e/0x1210000
    [ 6.091940] [00002107] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xaf9ad4d868ca/0x14623100 against cached device with sessionID/locationID 0x74352d40/0x14600000
    [ 6.091943] [00002107] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xaf9ad4d868ca/0x14623100 against cached device with sessionID/locationID 0x6a20b1bd/0x14400000
    [ 6.091946] [00002107] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xaf9ad4d868ca/0x14623100 against cached device with sessionID/locationID 0x653a4ce5/0x14300000
    [ 6.091949] [00002107] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xaf9ad4d868ca/0x14623100 against cached device with sessionID/locationID 0x5ce5b7d6/0x200000
    [ 6.091952] [00002107] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xaf9ad4d868ca/0x14623100 against cached device with sessionID/locationID 0x5523e666/0x1200000
    [ 6.091955] [00002107] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xaf9ad4d868ca/0x14623100 against cached device with sessionID/locationID 0x0/0x1000000
    [ 6.091958] [00002107] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xaf9ad4d868ca
    [ 6.092911] [00002107] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
    [ 6.092920] [00002107] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
    [ 6.092925] [00002107] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0200
    [ 6.092928] [00002107] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x00
    [ 6.092930] [00002107] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
    [ 6.092933] [00002107] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x00
    [ 6.092935] [00002107] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
    [ 6.092938] [00002107] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x05ac
    [ 6.092941] [00002107] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x12a8
    [ 6.092944] [00002107] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x1002
    [ 6.092946] [00002107] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
    [ 6.092948] [00002107] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
    [ 6.092951] [00002107] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x03
    [ 6.092953] [00002107] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x04
    ***[ void QUsbInfo::checkDevices() ]***
    [ 6.309304] [00000307] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
    [ 6.309317] [00000307] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
    [ 6.309323] [00000307] libusb: debug [handle_events] poll() 1 fds with timeout in 0ms
    [ 6.309339] [00000307] libusb: debug [handle_events] poll() returned 0
    [ 6.348891] [00002107] libusb: debug [darwin_check_configuration] active config: 0, first config: 1
[... 17 lines cut ...]
    [ 6.353512] [00002107] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xaf9ae460629d/0x14623100 against cached device with sessionID/locationID 0x88eac505/0x1211000
    [ 6.353515] [00002107] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xaf9ae460629d/0x14623100 against cached device with sessionID/locationID 0x7505644e/0x1210000
    [ 6.353518] [00002107] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xaf9ae460629d/0x14623100 against cached device with sessionID/locationID 0x74352d40/0x14600000
    [ 6.353520] [00002107] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xaf9ae460629d/0x14623100 against cached device with sessionID/locationID 0x6a20b1bd/0x14400000
    [ 6.353523] [00002107] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xaf9ae460629d/0x14623100 against cached device with sessionID/locationID 0x653a4ce5/0x14300000
    [ 6.353526] [00002107] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xaf9ae460629d/0x14623100 against cached device with sessionID/locationID 0x5ce5b7d6/0x200000
    [ 6.353528] [00002107] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xaf9ae460629d/0x14623100 against cached device with sessionID/locationID 0x5523e666/0x1200000
    [ 6.353531] [00002107] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xaf9ae460629d/0x14623100 against cached device with sessionID/locationID 0x0/0x1000000
[... cut ...]
    [ 6.354909] [00002107] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
    [ 6.354910] [00002107] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x03
    [ 6.354912] [00002107] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x05

Curiously, it seems that once it gets an error, it doesn't get any more events from any devices. So if I unplug my iPhone, then unplug my serial adapter, I don't see the serial adapter leave notification.

fpoussin, 2019-09-26:
I've never tested it on MacOS, so it might need some changes.
Have you checked permissions?

jhihn, 2019-09-27:
So I don't think this is anything related to your code, I think this is libusb. I just wanted to let you (and others) know it works (except for iPhones, apparently) aside from that one include problem. Not sure what that means for your project. Not sure why it didn't pick it up in the homebrew cellar.

jhihn, 2019-09-27:
Is there a ay to get the pid/vid without opening the device?

fpoussin, 2019-09-27:
I'm rewriting the callback code.
Detection does not rely on opening the devices.

jhihn, 2019-09-27:
Ah, ok.  I tried a naive hack:

    if (event == LIBUSB_HOTPLUG_EVENT_DEVICE_ARRIVED) {
        qDebug("hotplugCallback arrived: %04X:%04X", desc.idVendor, desc.idProduct);
        rc = libusb_open(device, &handle);
        if (rc != LIBUSB_SUCCESS) {
            qWarning("Could not open new USB device: %d", rc);
        }
        // Add to list
        dev.vid = desc.idVendor;
        dev.pid = desc.idProduct;
        device_list.append(dev);
        emit info->deviceInserted(device_list);
        return 0;
    }
But got: 
    Registering hotplug callback
    Starting...
    Press CTRL+C to close.
    hotplugCallback
    hotplugCallback removed: 05AC:12A8
    hotplugCallback
    hotplugCallback arrived: 05AC:12A8
    Could not open new USB device: -4
    devices inserted
    V05ac:P12a8
    hotplugCallback
    hotplugCallback removed: 05AC:12A8
    hotplugCallback
    hotplugCallback arrived: 05AC:12A8
    **libusb: warning [darwin_open] USBDeviceOpen: another process has device opened for exclusive access**
    devices inserted
    V05ac:P12a8

Not sure what the implications of that are?

fpoussin, 2019-09-27:
My guess is that the OS or an app is switching modes then opening it. This is what happens when an iphone is "trusted" on the host computer and you get access to the storage.

jhihn, 2019-09-27:
Thanks for being so responsive! Any idea when you'll have the new loop ready? I think my hack works for my purposes right now.

fpoussin, 2019-09-27:
Check the "callbacks" branch

fpoussin, 2019-10-01:
@jhihn did you get to try?

jhihn, 2019-10-01:
Not yet. Probably next week.

jhihn, 2019-10-10:
I'll have a go at it tomorrow/early next week. Sorry for the delay.

There was one other thing that I wanted to mention. And that is, for my needs, I changed the inserted/removed signals to provide the bus and port numbers. If at all possible, including that information (as well as the other USB fields) in the signal would be appreciated.

fpoussin, 2019-10-21:
@jhihn I've added code to report these as well as class and subclass.

jhihn, 2019-10-22:
Excited, excellent, will check it out tomorrow!

jhihn, 2019-10-22:
Yes, this is working most excellently! Thank you so much!

closed 2019-10-22 by jhihn

fpoussin, 2019-10-22:
I will merge, it will be in master/next release.

PR #35 (2019-10-22): Callbacks refactoring
Adding a callback to close devices when they are disconnected.
Cleaned QUsbInfo callback

Fixes #34