fpoussin/QtUsb #36: Is the Windows mingw version compilable?

angeluslove, 2019-11-07:
This project is great, but my software runs on mingw, and I hope this version will be available

fpoussin, 2019-11-07:
I've only tested with MSVC.
It previously relied on WDK so I could not use mingw, but it might work now.

closed 2019-11-17 by fpoussin

gtanisik, 2020-04-15:
It compiles with mingw with a small change in _src/usb/files.pri_ (line 56)

```
win32 {
    win32-g++{
        LIBS_PRIVATE += -ladvapi32 -lsetupapi
    }
    win32-msvc*{
        LIBS_PRIVATE += Advapi32.lib Setupapi.lib
    }
```

However i'm not able to install it as it gives the error:

```
...
D:\apps\QtUsb\build>mingw32-make install
cd src\ && ( if not exist Makefile C:\Qt\Qt5.14.1\5.14.1\mingw73_64\bin\qmake.exe -o Makefile D:\apps\QtUsb\src\src.pro ) && mingw32-make -f Makefile install
mingw32-make[1]: Entering directory 'D:/apps/QtUsb/build/src'
cd usb\ && ( if not exist Makefile C:\Qt\Qt5.14.1\5.14.1\mingw73_64\bin\qmake.exe -o Makefile D:\apps\QtUsb\src\usb\usb.pro ) && mingw32-make -f Makefile install
mingw32-make[2]: Entering directory 'D:/apps/QtUsb/build/src/usb'
copy /y ..\..\lib\Qt5Usb.dll C:\Qt\Qt5.14.1\5.14.1\mingw73_64\bin\Qt5Usb.dll
        1 file(s) copied.
C:\Qt\Qt5.14.1\5.14.1\mingw73_64\bin\qmake.exe -install sed  -e "s,D:/apps/QtUsb/build/lib,$$[QT_INSTALL_LIBS],g" -e "s,D:\\\\apps\\\\QtUsb\\\\build\\\\lib,$$[QT_INSTALL_LIBS],gi" -e "s,C:/Qt/Qt5.14.1/5.14.1/mingw73_64/lib,$$[QT_INSTALL_LIBS],g" -e "s,C:\\\\Qt\\\\Qt5.14.1\\\\5.14.1\\\\mingw73_64\\\\lib,$$[QT_INSTALL_LIBS],gi" ..\..\lib\Qt5Usb.prl > C:\Qt\Qt5.14.1\5.14.1\mingw73_64\lib\Qt5Usb.prl
C:\Qt\Qt5.14.1\5.14.1\mingw73_64\bin\qmake.exe -install sed  -e "s,D:/apps/QtUsb/build/lib,C:/Qt/Qt5.14.1/5.14.1/mingw73_64/lib,g" -e "s,D:\\\\apps\\\\QtUsb\\\\build\\\\lib,C:\\\\Qt\\\\Qt5.14.1\\\\5.14.1\\\\mingw73_64\\\\lib,gi" ..\..\build\lib\pkgconfig\Qt5Usb.pc > C:\Qt\Qt5.14.1\5.14.1\mingw73_64\lib\pkgconfig\Qt5Usb.pc
..\..\build\lib\pkgconfig\Qt5Usb.pc: No such file or directory
mingw32-make[2]: *** [Makefile:1487: install_target] Error 1
mingw32-make[2]: Leaving directory 'D:/apps/QtUsb/build/src/usb'
mingw32-make[1]: *** [Makefile:62: sub-usb-install_subtargets] Error 2
mingw32-make[1]: Leaving directory 'D:/apps/QtUsb/build/src'
mingw32-make: *** [Makefile:64: sub-src-install_subtargets] Error 2
```
I am not experienced with qt modules, so i couldn't find a solution yet. Therefore, i didn't open a pull request, as it's not a fully working solution.