fpoussin/QtUsb #39: it seems reverse between vid and pid

Quintin-Z, 2020-01-15:
QUsbDevice::Id usb_id;
    usb_id.pid = 0xABAB;
    usb_id.vid = 0xCDCD;

    m_usb_manager.addDevice(usb_id);

    QUsbDevice::IdList list = m_usb_manager.getPresentDevices();

    for(int i = 0 ; i < list.size() ; i++)
    {
         qInfo("V:%04x:P:%04x", list.at(i).vid, list.at(i).pid);
//        qDebug()<<"vid:"<<list.at(i).vid<<"pid:"<<list.at(i).pid;
    }

//////////////////////////////////////////
    usb_id.pid = 0xABAB;
    usb_id.vid = 0xCDCD;

   is correct pid and vid，but the code run can‘t find the usb device，but if

    usb_id.pid = 0xCDCD;
    usb_id.vid = 0xABAB;

  that can get present deevcices

fpoussin, 2020-01-15:
Which version and platform are you using?

Quintin-Z, 2020-01-16:
winodws 10,QT:5.12.6,QtUsb:v0.5.1, vs2017
QtUsb lib is build by myself

Quintin-Z, 2020-01-16:
and is the lib can use hid usb device?

Quintin-Z, 2020-01-16:
will it can use hid device in the future?

i use “UsbNotifications” example code ，my usb device vid and pid can be found。
but i cannot understand the getPresentDevices API, that seems to return a list of all usb device.
but in fact that is not true,so i so confuse 

can you please provide more example or document to me？

thinks a lot. the QtUsb lib is very potential lib，hope better and better.

fpoussin, 2020-01-16:
HID is not planned, it only does raw usb. You can implement HID on top of QtUsb.
getPresentDevices as the name suggests returns all usb devices present and accessible on the OS.
If a driver already captured the device, you will not see it and that's normal.

Documentation is here: https://fpoussin.github.io/doxygen/qtusb/0.5.x/qtusb-module.html

Quintin-Z, 2020-01-17:
hello fpoussin,
i want to implement HID on top of QtUsb, but i am read the code found if not modify source of QtUsb, i can not get the usb  descriptor,because a lot of infomation of struct libusb_device_descriptor of libusb is cut in QtUsb. I am confuse how to implement a usb class on the top of QtUsb.

Can you offer an example of some usb class implement top of QtUsb??
or other more example except repo of QtUsb in github.
thanks!

fpoussin, 2020-01-17:
Can you test the latest beta release? it exposes more descriptor properties.
I have no extra examples using these.

I will try to make a basic hid implementation.

closed 2020-01-30 by fpoussin

fpoussin, 2020-02-01:
You can check the latest beta release, I've added HID.