fpoussin/QtUsb #43: Weird behavior on Raspberry Pi4

jhihn, 2020-02-24:
**Describe the bug**
Only on Raspberry Pi (4), the inserted VID:PID is often wrong. The same code works fine on OSX, Linux. Usually the first insert after invoking the application is correct. What then happens is the VID:PID on removal is not the same. Subsequent inserts and removals are 

**PLatform:**
 - OS: Raspbian  ( https://www.raspberrypi.org/downloads/raspbian/ )
 - Version 2020-02-13

**To Reproduce**
Steps to reproduce the behavior:
Compile as usual, monitor for insertion and removal events.
Insert a device of a known VID:PID. Remove device. Observe the right VID:PID was printed. Repeat several times.

**Expected behavior**
The proper VID:PID will always be shown.

**Additional context**
I'm using a iPhone with 05AC:12A8. Log is of me inserting the device into various ports  of a hub. The iPhone's VIDPID is 05AC:12A8
[insert.txt](https://github.com/fpoussin/QtUsb/files/4245504/insert.txt)

jhihn, 2020-02-24:
Linux kernel reports the right VID:PID

fpoussin, 2020-02-24:
Are you using code on master branch or a release?

fpoussin, 2020-02-24:
Try again with latest master, I had made a mistake when cleaning the code.

jhihn, 2020-02-24:
Nice. Looks fixed.

closed 2020-02-24 by fpoussin