fpoussin/QtUsb #45: Error during installation of the Qt USB library  - returned code "0x2"

serj053, 2020-03-13:
**Describe the bug**
A clear and concise description of what the bug is.-Okay, I'll try.

Friends, help me figure out the error that occurs when installing the Qt USB library . this error occurs when executing the command - " build_msvc2017.bat x64 G:\Qt\5.12.3\msvc2017_64". The Visual Studio 2019 command line is used . Here it is the error - ```
NMAKE : fatal error U1077: " C:\Program Files (x86)\Microsoft Visual Studio\2019\ 
Enterprise\VC\Tools\MSVC\14.24.28314\bin\Hostx64\x64\cl.EXE": returned code
"0x2"
```
This error occurs several times.

**PLatform:**
 - OS: [e.g. Ubuntu]
 - Version [e.g. 18.04]

64 bit operating system,
Windows 7, Visual Studio  community 2019,
Separate component for VS:
NET Framework 4.6.1 Targeting Pack,
SDK  Framework  4.8,
IntelliCode,
Live Share,
C++ Clang for Windows,
C++ modules for build tools version 142 (x64 or x86 experimental),
Update for the distributed C++ component 2019,
SDK for universal CRT for Windows,
C++/CL support for build tools version 142 (14.24),
CMake C++ tools for Windows,
C++ Clang-cl for build tools version 142(x64/x86),
IncrediBuild - faster build times,
MSVC versions-140 C++VS 2015 build tools (versions 14.00),
MSVC versions-142 libraries C++VS 2019 for x68/x86, eliminating Spectre risks (version 14.24),
MSVC versions-142 C++VS 2019 build tools (versions 14.00), for x68  or x86,
Web deployment,
Tools related to connecting and publishing,
C++ profiling tools,
C++ Address Sanitizer,
IntelliTrace,
Jit debugger,
Package SDK for TypeScript 3.7,
Package SDK for Windows 10 (10.0.16299.0),
Package SDK for Windows 10 (10.0.17134.0),
Package SDK for Windows 10 (10.0.18362.0),
ATL library for the latest version of Build Tools v 142 (x86 or x64),
MFC library for the latest version of Build Tools v 142 (x86 or x64)
Package manager NuGet,
Code map,
Class designer,
Editor DGML?
Windows driver KIT.

Qt Creator 4.11.1.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

All the way, just in case:

Path=C:\Perl64\site\bin;C:\Perl64\bin;C:\Program Files (x86)\Common Files\Oracle\Java\jav
apath;C:\ProgramData\Oracle\Java\javapath;C:\Omega\SoapySDR_win64_install;C:\Program File
s\CMake\bin\;C:\Windows\System32;C:\Windows\SysWOW64\WindowsPowerShell\v1.0;C:\Omega\Soap
ySDR_win64_install\bin\;C:\Omega\CubicSDR_win64\x64\modules\;C:\Windows;C:\Program Files
(x86)\GnuWin32\bin;C:\SysGCC\mingw64\bin;C:\Program Files\PothosSDR\lib\SoapySDR\modules0
.8;G:\MATLAB\R2019b\runtime\win64;G:\MATLAB\R2019b\bin;G:\MATLAB\R2019b\polyspace\bin;C:\
Program Files\Git\cmd;C:\Program Files\bladeRF\x64;C:\Program Files\GTK2-Runtime Win64\bi
n;C:\boost_1_72_0;C:\Boost\boost_1_72_0\libs;C:\Program Files\bladeRF\matlab;G:\Premake;G
:G:\test\wxWidgets-3.1.3\include;G:\test\wxWidgets-3.1.3\lib;G:\test\wxWidgets\include;C:
\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.24.28314\bi
n\Hostx64\x64;C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\MSBuild\Curr
ent\Bin;
PATHEXT=.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context about the problem here.

Code snippets with errors.

This message appears several times- "The system cannot find the specified path."but it seemed to me that it does not affect the process, maybe I'm wrong.

`G:\QtUsb>build_msvc2017.bat x64 G:\Qt\5.12.3\msvc2017_64
The system cannot find the specified path.
Microsoft (R) Build Engine версии 16.4.0+e901037fe для .NET Framework
(C) Корпорация Майкрософт (Microsoft Corporation). Все права защищены.

Build started 13.03.2020 12:48:20.
Проект "G:\QtUsb\libusb\msvc\libusb_static_2017.vcxproj" в узле 1 (default targets).
C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\MSBuild\Microsof
t\VC\v150\Microsoft.CppBuild.targets(402,5): warning MSB8038: Spectre risk elimination is enabled , but no Spectre libraries with eliminated risks were found.

`G:\Qt\5.12.3\msvc2017_64\include\QtCore\qglobal.h(45): fatal error C1083: Не уда
ется открыть файл включение: type_traits: No such file or directory,
NMAKE : fatal error U1077: "C:\Program Files (x86)\Microsoft Visual Studio\2019\
Enterprise\VC\Tools\MSVC\14.24.28314\bin\Hostx64\x64\cl.EXE" : возвращенный код
"0x2"
Stop.
NMAKE : fatal error U1077: "C:\Program Files (x86)\Microsoft Visual Studio\2019\
Enterprise\VC\Tools\MSVC\14.24.28314\bin\Hostx64\x64\nmake.exe" : returned code "0x2"
Stop.
NMAKE : fatal error U1077: cd :returned code "0x2"

`ils\postgresql\pgsql\include -IG:\Qt\5.12.3\msvc2017_64\mkspecs\win32-msvc -TP .
.\..\include\QtUsb\QtUsbDepends
QtUsbDepends
G:\Qt\5.12.3\msvc2017_64\include\QtCore\qglobal.h(45): fatal error C1083: Не уда
ется открыть файл включение: type_traits: No such file or directory,
NMAKE : fatal error U1077: "C:\Program Files (x86)\Microsoft Visual Studio\2019\
Enterprise\VC\Tools\MSVC\14.24.28314\bin\Hostx64\x64\cl.EXE" : returned code
Stop.
NMAKE : fatal error U1077: "C:\Program Files (x86)\Microsoft Visual Studio\2019\
Enterprise\VC\Tools\MSVC\14.24.28314\bin\Hostx64\x64\nmake.exe" : returned code "0x2"
Stop.
NMAKE : fatal error U1077: cd : returned code "0x2"
Stop.
NMAKE : fatal error U1077: cd : returned code "0x2"
Stop.

G:\QtUsb>`
Stop.
NMAKE : fatal error U1077: cd :returned code "0x2"
Stop.
The system cannot find the specified path."
`
I hope for your help. Sincerely Sergey
((I'm waiting for an answer like a bird of summer)))

fpoussin, 2020-03-13:
You are using the msvc2017 script with msvc2019.
The script already calls the 2017 environment, so this won't work.
I will make a script for msvc2019.

closed 2020-03-13 by fpoussin

serj053, 2020-03-13:
You answered so quickly that I jumped up in my chair with pleasure , and you can ask how your plans are for time to finalize the library, or it makes sense to reinstall VS 2019 on VS 2017, I really want to make friends with one USB device .))

serj053, 2020-03-13:
> You are using the msvc2017 script with msvc2019.
> The script already calls the 2017 environment, so this won't work.
> I will make a script for msvc2019.

You answered so quickly that I jumped up in my chair with pleasure , and you can ask how your plans are for time to finalize the library, or it makes sense to reinstall VS 2019 on VS 2017, I really want to make friends with one USB device .))

fpoussin, 2020-03-13:
You can use msvc 2019, just use the update script.

```
build_msvc.bat 2019 x64 G:\Qt\5.12.3\msvc2017_64
```

serj053, 2020-03-13:
> You answered so quickly that I jumped up in my chair with pleasure , and you can ask how your plans are for time to finalize the library, or it makes sense to reinstall VS 2019 on VS 2017, I really [...]

Fabien, thank you , I built the library using your new link, the errors are preserved , but there are changes for VS 2017 output lines 234, "NMAKE :fatal error U1077" errors -8, " qglobal.h(45): fatal error C1083: failed you can open the" errors - 2 " file. For the VS 2019 variant, the output lines are 294, "NMAKE : fatal error U1077" errors are 8 , and " qglobal.h(45): fatal error C1083: unable to open the file " errors - 4.
Despite the errors, I tried to build libusb and hidapi in VS 2019 using files with the "sin" extension in each directory, the operations passed without errors, does this mean that I can use these libraries?, and also could you give a link to the material that will allow me to understand as a novice programmer how to use QtUSB with Qt Creator and not bother respected people with stupid questions.
Sincerely, Sergey.

fpoussin, 2020-03-13:
You can have a look at this automated build log: https://ci.appveyor.com/project/fpoussin/qtusb/builds/31458448/job/5ucuikf8thhxeiqw

I have also updated the msvc build script, it works for me using msvc 2019 and W10 SDK 10.0.16299.0

serj053, 2020-03-15:
> You can have a look at this automated build log: https://ci.appveyor.com/project/fpoussin/qtusb/builds/31458448/job/5ucuikf8thhxeiqw
> 
> I have also updated the msvc build script, it works for me using msvc 2019 and W10 SDK 10.0.16299.0
I have here snow outside the window is falling.
Thank you all looked and tried again, but...
 , so I wanted to write, hurray everything turned out !!! , but not fate, throws an error that "'qmake" is not an internal or external command executed by a program or batch file.", although I am a file "qmake .exe"  put directly in the QtUSB folder where the file is located - " build_msvc.bat ", why can it be? maybe there's a trick here that beginners don't know?, although when I execute this file-"qmake.exe" it is executed., dark forest. 
With respect and undying hope, Sergey.

fpoussin, 2020-03-15:
I have updated the scripts again, now you don't need to build libusb or hidapi anymore, they are built with QtUsb. You can build everything with Qt creator, or build_msvc.bat