fpoussin/QtUsb #48: OSX: Project ERROR: Could not find hidapi-libusb using PKGCONFIG

jhihn, 2020-03-31:
**Describe the bug**
When doing a make, I get
Project ERROR: Could not find hidapi-libusb using PKGCONFIG
However hidapi and libusb are installed. There is no brew formula for hidapi-libusb

**PLatform:**
 - OS: MacOS
 - Version:Catalina

**To Reproduce**
Steps to reproduce the behavior:
git clone https://github.com/fpoussin/QtUsb.git
cd QtUsb
mkdir build && cd build
make

**Expected behavior**
It shoudl compile

**Additional context**
It fails with 
Project MESSAGE: Build as Qt module
Project ERROR: Could not find hidapi-libusb using PKGCONFIG
make[1]: *** [sub-usb-make_first] Error 3
make: *** [sub-src-make_first] Error 2

fpoussin, 2020-04-01:
Can you list installed lib packages with ```pkg-config --list-all```?

You can try to edit https://github.com/fpoussin/QtUsb/blob/master/src/usb/files.pri#L110 to change *hidapi-libusb* to *hidapi*

jhihn, 2020-04-02:
That seems to have fixed it! Thanks!

closed 2020-04-03 by fpoussin