fpoussin/QtUsb #50: QT modules usb not found

joejond, 2020-04-28:
I'm using ubuntu 18.04, and want to try using Qt USB by adding repo to my ubuntu.
I'm succeed install in my pc using apt install libqt5usb5 and libqt5usb5-dev.

after install I want to try this module to my QT  (qt5.9), unfortunately this error come up. 
"Unknown module(s) in QT: usb"

How to check in my pc, is my  module installed correctly?

fpoussin, 2020-04-28:
I haven't updated the packages to the latest version, but it should still work.
Are you using the system Qt version and not an official package?

fpoussin, 2020-04-28:
I've updated the packages to the latest release

joejond, 2020-04-29:
Thanks for your update.. currently I using include in project .pro file, and its working.
If you update with newer version, I will try again using apt install. 

Is there any sample creating protocol using pc host and device using usb?

Verywell thanks

fpoussin, 2020-04-29:
As I said, package have been updated, you can retry.

There are no examples of protocols as it's out of the scope of QtUsb.
You can have a look at modbus: https://doc.qt.io/qt-5/qtmodbus-backends.html

closed 2020-05-07 by fpoussin

selmehtar, 2020-05-22:
I have the same issue using Windows. I added the library to my project and added QT += usb to the .pro file and get the error "Project ERROR: Unknown module(s) in QT: usb".

fpoussin, 2020-05-22:
This is normal, please read the instructions