fpoussin/QtUsb #51: can't r get usb input data by QUsbEndpoint::interruptEndpoint

gleke, 2020-05-13:
**Describe the bug**
A clear and concise description of what the bug is.
void UsbExample::setupDevice()
{
    /* There are 2 ways of identifying devices depending on the platform.
   * You can use both methods, only one will be taken into account.
   */

    printf("\nsetupDevice");

    m_usb_dev->setLogLevel(QUsbDevice::logDebug);

    //脉搏仪
    m_filter.pid = 0x2007;
    m_filter.vid = 0x5131;

    //
    m_config.alternate = 0;
    m_config.config = 1;
    m_config.interface = 0;

    //
    m_usb_dev->setId(m_filter);
    m_usb_dev->setConfig(m_config);

}
bool UsbExample::openDevice()
{
    qDebug("Opening");
    printf("\nOpening");

    if (m_usb_dev->open() == QUsbDevice::statusOK) {
        // Device is open
        return  this->openHandle();
    }
    return false;
}

bool UsbExample::openHandle()
{
    printf("\nOpening Handle");
    bool a = false, b = false;
    m_read_ep = new QUsbEndpoint(m_usb_dev, QUsbEndpoint::interruptEndpoint, USB_ENDPOINT_IN);
    //m_write_ep = new QUsbEndpoint(m_usb_dev, QUsbEndpoint::bulkEndpoint, USB_ENDPOINT_OUT);

    connect(m_read_ep, SIGNAL(readyRead()), this, SLOT(onReadyRead()));
    //connect(m_write_ep, SIGNAL(bytesWritten(qint64)), this, SLOT(onWriteComplete(qint64)));

    a = m_read_ep->open(QIODevice::ReadOnly);
    if (a) {
        m_read_ep->setPolling(true);
    }

//    b = m_write_ep->open(QIODevice::WriteOnly);

//    printf("\n===m_read_ep.type %d",m_read_ep->m_type);
    return a;
    //return a && b;
}

void UsbExample::onReadyRead()
{
    printf("\nonReadyRead");
    this->read(&m_recv);

   // this->write(&m_send);
}

**PLatform:**
 - OS: [win7]
 - Version [windows7]
-qt 5.12.7 mingw64

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context about the problem here.

gleke, 2020-05-13:
Parameter | Value
-- | --
Hardware ID | USB\VID_5131&PID_2007&REV_0000
Setup Class | HIDClass
Class GUID | {745a17a0-74d3-11d0-b6fe-00a0c90f57da}
PDO Name | \Device\00000087
Service Name | HidUsb

gleke, 2020-05-13:
Interface Descriptor
--
bLength | 0x9
bInterfaceNumber | 0x0
bAlternateSetting | 0x0
bNumEndpoints | 0x2
bInterfaceClass | 0x3 (Human Interface  Device)
bInterfaceSubClass | 0x0 (No Subclass)
bInterfaceProtocol | 0x0 (None)
iInterface | 0x5

gleke, 2020-05-13:
Endpoint Descriptor
--
bLength | 0x7
bEndpointAddress | 0x82 [IN]
bmAttributes | 0x3  (USB_ENDPOINT_TYPE_INTERRUPT)
wMaxPacketSize | 0x8
bInterval | 0x8

gleke, 2020-05-13:
Endpoint Descriptor
--
bLength | 0x7
bEndpointAddress | 0x2 [OUT]
bmAttributes | 0x3  (USB_ENDPOINT_TYPE_INTERRUPT)
wMaxPacketSize | 0x8
bInterval | 0x8

gleke, 2020-05-13:
Parameter | Value
-- | --
Configuration Descriptor
bLength | 0x9
bDescriptorType | USB_CONFIGURATION_DESCRIPTOR_TYPE
wTotalLength | 0x29
bNumInterfaces | 0x1
iConfiguration | 0x4
bmAttributes | 0xA0 ( Bus_Powered  Remote_Wakeup )
MaxPower | 0x23

closed 2020-05-13 by gleke

fpoussin, 2020-05-13:
This is a HID device, so you need to use the QHidDevice class.

gleke, 2020-05-13:
QHidDevice class  can't  use QUsbEndpoint::interruptEndpoint???

fpoussin, 2020-05-13:
It's abstracted by the OS, you just have to use the read() and write() methods.

gleke, 2020-05-13:
Thanks!
     but my usb device  only support  Interrupt transfer, I can't write() methods

fpoussin, 2020-05-13:
Did you try?

gleke, 2020-05-13:
thanks for you !

I can't get usb data!

HidExample::HidExample(QObject *parent)     : QObject(parent), m_hid_dev(new QHidDevice()) {     m_send.append(static_cast<char&gt;(0xF1u));     m_send.append(static_cast<char&gt;(0x80u));     timer=new QTimer(); 
     QObject::connect(timer, SIGNAL(timeout()), this, SLOT(getData()));     timer-&gt;start(200); 
     if (this-&gt;openDevice()) {         qInfo("Device open!");         printf("Device open!");         this-&gt;write(&amp;m_send);     } else {         qWarning("Could not open device!");     } } 
 HidExample::~HidExample() {     this-&gt;closeDevice();     m_hid_dev-&gt;deleteLater(); } 
 bool HidExample::openDevice() {     qDebug("Opening"); 
     return m_hid_dev-&gt;open(0x5131,0x2007); } 
 void HidExample::closeDevice() {     qDebug("Closing"); 
     if (m_hid_dev-&gt;isOpen()) {         m_hid_dev-&gt;close();     } } QString HidExample::getData() {     printf("===========open result========= ");     QByteArray array;     read(&amp;array);     printf("======array.count%d",array.count());     return "asd"; } void HidExample::read(QByteArray *buf) {     QByteArray b(128, '0');     m_hid_dev-&gt;read(&amp;b, b.size(),1);     printf("Reading");     qDebug() << "Reading" << b << b.size();     buf-&gt;append(b); } 
 void HidExample::write(QByteArray *buf) {     qDebug() << "Writing" << *buf << buf-&gt;size();     if (m_hid_dev-&gt;write(buf, buf-&gt;size()) < 0) {         qWarning("write failed");     } }

gleke, 2020-05-13:
Tanks you!
 the bus hound can get my usb input data,please see the png .

gleke, 2020-05-13:
Thanks !
 these  pngs of  bus hound can help understand Interrupt transfer

leifliddy, 2021-12-23:
I think this would be way easier with python.

Both of these projects support variants of the  ```5131:2007``` USB HID device.
https://github.com/Progman2k/USB_Watchdog
https://github.com/leifliddy/usb-watchdog