fpoussin/QtUsb #52: add getReport(size_t length, int timeout_ms) and sendReport(const QByteArray &data)

gleke, 2020-05-13:
**Is your feature request related to a problem? Please describe.**
int QHIDevice::sendReport(const QByteArray &data) {
    if (_device != nullptr) return -1;
    return hid_send_feature_report(_device,reinterpret_cast<const unsigned char*>(data.constData()),data.size());
}

void QHIDevice::getReport(size_t length, int timeout_ms) {
	if (_device == nullptr) return;
	_timeout.start(timeout_ms);
    expectData = length;
    recvBuf.clear();
    poll.start();
}
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

fpoussin, 2020-05-13:
I have added these to the master branch, give it a try.

gleke, 2020-05-14:
my hid usb device use Interrupt transfer,but I can't get data by :getFeatureReport(QByteArray *buf)  or read(QByteArray *buf)  but the bus hound tool can

```
HidExample::HidExample(QObject *parent)
    : QObject(parent), m_hid_dev(new QHidDevice())
{
    m_send.append(static_cast<char>(0xF1u));
    m_send.append(static_cast<char>(0x80u));
    timer=new QTimer();

    QObject::connect(timer, SIGNAL(timeout()), this, SLOT(getData()));
    timer->start(200);

    if (this->openDevice()) {
        qInfo("Device open!");
        printf("Device open!");
        this->write(&m_send);
    } else {
        qWarning("Could not open device!");
    }
}

void HidExample::getData()
{

    QByteArray array;
    getFeatureReport(&array);
    
    for(int i=0;i<array.size();i++){
        char ch = array[i] ;
        int value = int(ch);
        printf("\n=======getData value:=%d",value);
    }

}
void HidExample::getFeatureReport(QByteArray *buf){
    QByteArray b(128, '0');
    m_hid_dev->getFeatureReport(&b, b.size());
    buf->append(b);

}
```
```
HidExample::HidExample(QObject *parent)
    : QObject(parent), m_hid_dev(new QHidDevice())
{
    m_send.append(static_cast<char>(0xF1u));
    m_send.append(static_cast<char>(0x80u));
    timer=new QTimer();

    QObject::connect(timer, SIGNAL(timeout()), this, SLOT(getData()));
    timer->start(200);

    if (this->openDevice()) {
        qInfo("Device open!");
        printf("Device open!");
        this->write(&m_send);
    } else {
        qWarning("Could not open device!");
    }
}

void HidExample::getData()
{

    QByteArray array;
    read(&array);
    
    for(int i=0;i<array.size();i++){
        char ch = array[i] ;
        int value = int(ch);
        printf("\n=======getData value:=%d",value);
    }

}
void HidExample::read(QByteArray *buf)
{
    QByteArray b(128, '0');

    m_hid_dev->blockSignals(false);
    m_hid_dev->read(&b, b.size(),1);

    qDebug() << "Reading" << b << b.size();
    buf->append(b);
}
```

gleke, 2020-05-14:
```
Bus Hound 6.01 capture on Windows Vista Service Pack 1 (x64). Complements of www.perisoft.net

www

  Device - Device ID (followed by the endpoint for USB devices)
            (12) USB input device
            (15) HID-compliant device
  Address - FireWire async address or channel number
  Length - Total transfer length
  Phase  - Phase Type
            IN    Data in transfer           
  Data   - Hex dump of the data transferred
  Descr  - Description of the phase
  Cmd... - Position in the captured data

Device  Address              Length    Phase  Data  Description       Cmd.Phase.Ofs(rep)
------  -------------------  --------  -----  ----  ----------------  ------------------
  12.2                              1  IN     c2    .                   258361.1.0        
  12.2                              1  IN     c1    .                   258362.1.0        
  12.2                              1  IN     c0    .                   258363.1.0        
  12.2                              1  IN     bf    .                   258364.1.0        
  12.2                              1  IN     be    .                   258365.1.0        
  12.2                              1  IN     bd    .                   258366.1.0        
  12.2                              1  IN     bc    .                   258367.1.0        
  12.2                              1  IN     bb    .                   258368.1.0        
  12.2                              1  IN     ba    .                   258369.1.0        
  12.2                              1  IN     b9    .                   258370.1.0        
  12.2                              1  IN     b7    .                   258371.1.0        
  12.2                              1  IN     b6    .                   258372.1.0(2)     
  12.2                              1  IN     b4    .                   258374.1.0        
  12.2                              1  IN     b3    .                   258375.1.0        
  12.2                              1  IN     b2    .                   258376.1.0(2)     
  12.2                              1  IN     b1    .                   258378.1.0        
  12.2                              1  IN     b0    .                   258379.1.0        
  12.2                              1  IN     af    .                   258380.1.0        
  12.2                              1  IN     ae    .                   258381.1.0        
  12.2                              1  IN     ad    .                   258382.1.0(2)     
  12.2                              1  IN     ac    .                   258384.1.0        
  12.2                              1  IN     ab    .                   258385.1.0        
  12.2                              1  IN     aa    .                   258386.1.0(2)     
  12.2                              1  IN     a9    .                   258388.1.0        
  12.2                              1  IN     a8    .                   258389.1.0(2)     
  12.2                              1  IN     a7    .                   258391.1.0(2)     
  12.2                              1  IN     a6    .                   258393.1.0(2)     
  12.2                              1  IN     a5    .                   258395.1.0(3)     
  12.2                              1  IN     a4    .                   258398.1.0(2)     
  12.2                              1  IN     a3    .                   258400.1.0(4)     
  12.2                              1  IN     a2    .                   258404.1.0(2)     
  12.2                              1  IN     a1    .                   258406.1.0(2)     
  12.2                              1  IN     a0    .                   258408.1.0(3)     
  12.2                              1  IN     9f    .                   258411.1.0        
  12.2                              1  IN     9e    .                   258412.1.0(2)     
  12.2                              1  IN     9d    .                   258414.1.0(2)     
  12.2                              1  IN     9c    .                   258416.1.0        
  12.2                              1  IN     9b    .                   258417.1.0(2)     
  12.2                              1  IN     9a    .                   258419.1.0        
  12.2                              1  IN     99    .                   258420.1.0(2)     
  12.2                              1  IN     98    .                   258422.1.0        
  12.2                              1  IN     97    .                   258423.1.0(2)     
  12.2                              1  IN     96    .                   258425.1.0(2)     
  12.2                              1  IN     95    .                   258427.1.0(3)     
  12.2                              1  IN     94    .                   258430.1.0(10)    
  12.2                              1  IN     95    .                   258440.1.0(5)     
  12.2                              1  IN     96    .                   258445.1.0(4)     
  12.2                              1  IN     97    .                   258449.1.0(3)     
  12.2                              1  IN     98    .                   258452.1.0(3)     
  12.2                              1  IN     99    .                   258455.1.0(5)     
  12.2                              1  IN     9a    .                   258460.1.0(4)     
  12.2                              1  IN     9b    .                   258464.1.0(5)     
[... cut ...]
  12.2                              1  IN     9d    .                   258474.1.0(6)     
  12.2                              1  IN     9e    .                   258480.1.0(8)     
  12.2                              1  IN     9f    .                   258488.1.0(10)    
  12.2                              1  IN     a0    .                   258498.1.0(21)    
  12.2                              1  IN     9f    .                   258519.1.0(13)    
  12.2                              1  IN     9e    .                   258532.1.0(9)  
```

gleke, 2020-05-14:
my hid usb device use Interrupt transfer,but I can't get data by :getFeatureReport(QByteArray *buf)  or read(QByteArray *buf)  but the bus hound tool can

```
HidExample::HidExample(QObject *parent)
    : QObject(parent), m_hid_dev(new QHidDevice())
{
    m_send.append(static_cast<char>(0xF1u));
    m_send.append(static_cast<char>(0x80u));
    timer=new QTimer();

    QObject::connect(timer, SIGNAL(timeout()), this, SLOT(getData()));
    timer->start(200);

    if (this->openDevice()) {
        qInfo("Device open!");
        printf("Device open!");
        this->write(&m_send);
    } else {
        qWarning("Could not open device!");
    }
}

void HidExample::getData()
{

    QByteArray array;
    getFeatureReport(&array);
    
    for(int i=0;i<array.size();i++){
        char ch = array[i] ;
        int value = int(ch);
        printf("\n=======getData value:=%d",value);
    }

}
void HidExample::getFeatureReport(QByteArray *buf){
    QByteArray b(128, '0');
    m_hid_dev->getFeatureReport(&b, b.size());
    buf->append(b);

}
```
```
HidExample::HidExample(QObject *parent)
    : QObject(parent), m_hid_dev(new QHidDevice())
{
    m_send.append(static_cast<char>(0xF1u));
    m_send.append(static_cast<char>(0x80u));
    timer=new QTimer();

    QObject::connect(timer, SIGNAL(timeout()), this, SLOT(getData()));
    timer->start(200);

    if (this->openDevice()) {
        qInfo("Device open!");
        printf("Device open!");
        this->write(&m_send);
    } else {
        qWarning("Could not open device!");
    }
}

void HidExample::getData()
{

    QByteArray array;
    read(&array);
    
    for(int i=0;i<array.size();i++){
        char ch = array[i] ;
[... 69 lines cut ...]
  12.2                              1  IN     9d    .                   258414.1.0(2)     
  12.2                              1  IN     9c    .                   258416.1.0        
  12.2                              1  IN     9b    .                   258417.1.0(2)     
  12.2                              1  IN     9a    .                   258419.1.0        
  12.2                              1  IN     99    .                   258420.1.0(2)     
  12.2                              1  IN     98    .                   258422.1.0        
  12.2                              1  IN     97    .                   258423.1.0(2)     
  12.2                              1  IN     96    .                   258425.1.0(2)     
  12.2                              1  IN     95    .                   258427.1.0(3)     
  12.2                              1  IN     94    .                   258430.1.0(10)    
  12.2                              1  IN     95    .                   258440.1.0(5)     
  12.2                              1  IN     96    .                   258445.1.0(4)     
  12.2                              1  IN     97    .                   258449.1.0(3)     
  12.2                              1  IN     98    .                   258452.1.0(3)     
  12.2                              1  IN     99    .                   258455.1.0(5)     
  12.2                              1  IN     9a    .                   258460.1.0(4)     
  12.2                              1  IN     9b    .                   258464.1.0(5)     
  12.2                              1  IN     9c    .                   258469.1.0(5)     
  12.2                              1  IN     9d    .                   258474.1.0(6)     
  12.2                              1  IN     9e    .                   258480.1.0(8)     
  12.2                              1  IN     9f    .                   258488.1.0(10)    
  12.2                              1  IN     a0    .                   258498.1.0(21)    
  12.2                              1  IN     9f    .                   258519.1.0(13)    
  12.2                              1  IN     9e    .                   258532.1.0(9)  
```

gleke, 2020-05-14:
Device - Device ID (followed by the endpoint for USB devices)
 (12) USB input device 
I use  QUsbDevice ,but can't capture  usb device data like Bus Hound 6.01 
```
UsbExample::UsbExample(QObject *parent)
    : QObject(parent), m_usb_dev(new QUsbDevice()), m_read_ep(Q_NULLPTR), m_write_ep(Q_NULLPTR)
{
    this->setupDevice();

    if (this->openDevice()) {
        printf("\nDevice open!");
        //this->write(&m_send);      

    } else {
        qWarning("Could not open device!");
    }
  
}

UsbExample::~UsbExample()
{
    this->closeDevice();
    m_usb_dev->deleteLater();
}

void UsbExample::setupDevice()
{
    /* There are 2 ways of identifying devices depending on the platform.
   * You can use both methods, only one will be taken into account.
   */

    printf("\nsetupDevice");

    m_usb_dev->setLogLevel(QUsbDevice::logDebug);

    //脉搏仪
    m_filter.pid = 0x2007;
    m_filter.vid = 0x5131;

    //
    m_config.alternate = 0;
    m_config.config = 1;
    m_config.interface = 0;

    //
    m_usb_dev->setId(m_filter);
    m_usb_dev->setConfig(m_config);

}
bool UsbExample::openDevice()
{
    qDebug("Opening");
    printf("\nOpening");

    if (m_usb_dev->open() == QUsbDevice::statusOK) {
        // Device is open
        return  this->openHandle();
    }
    return false;
}

bool UsbExample::openHandle()
{
    printf("\nOpening Handle");
    bool a = false, b = false;

    m_read_ep = new QUsbEndpoint(m_usb_dev, QUsbEndpoint::interruptEndpoint, USB_ENDPOINT_IN);
    //m_write_ep = new QUsbEndpoint(m_usb_dev, QUsbEndpoint::bulkEndpoint, USB_ENDPOINT_OUT);

    connect(m_read_ep, SIGNAL(readyRead()), this, SLOT(onReadyRead()));
    //connect(m_write_ep, SIGNAL(bytesWritten(qint64)), this, SLOT(onWriteComplete(qint64)));

    a = m_read_ep->open(QIODevice::ReadOnly);
    if (a) {
        m_read_ep->setPolling(true);
    }

//    b = m_write_ep->open(QIODevice::WriteOnly);

//    printf("\n===m_read_ep.type %d",m_read_ep->m_type);
    return a;
    //return a && b;
}
void UsbExample::read(QByteArray *buf)
{
    QByteArray b(m_read_ep->readAll());
    printf("\nReading ,%d",b.size());
    for(int i=0;i<b.size();i++){
        char ch = b[i] ;
        int value = int(ch);
        printf("\n=======value%d",value);
    }
    qDebug() << "Reading" << b << b.size();
    buf->append(b);
}

void UsbExample::write(QByteArray *buf)
{
    qDebug() << "Writing" << *buf << buf->size();
    if (m_write_ep->write(buf->constData(), buf->size()) < 0) {
        qWarning("write failed");
    }
}

void UsbExample::onReadyRead()
{
    printf("\nonReadyRead");
    this->read(&m_recv);

   // this->write(&m_send);
}
```

gleke, 2020-05-14:
```
#ifndef USBEXAMPLE_H
#define USBEXAMPLE_H

#include <QObject>
#include "qusbdevice.h"
#include "qusbendpoint.h"
#include "qhiddevice.h"
#include <QDebug>
#include <QTimer>
const quint8 USB_ENDPOINT_IN = 0x02; /* Bulk output endpoint for responses */
const quint8 USB_ENDPOINT_OUT = 0x02; /* Bulk input endpoint for commands */
const quint16 USB_TIMEOUT_MSEC = 300;

class UsbExample : public QObject
{
    Q_OBJECT
public:
    explicit UsbExample(QObject *parent = Q_NULLPTR);
    ~UsbExample(void);
    void setupDevice(void);
    bool openDevice(void);
    void closeDevice(void);
    bool openHandle(void);
    void closeHandle(void);
    void read(QByteArray *buf);
    void write(QByteArray *buf);

public slots:
    void onReadyRead(void);
    void onWriteComplete(qint64 bytes);

    void onT_timerOut();

signals:

private:
    QUsbDevice *m_usb_dev;

    QUsbEndpoint *m_read_ep, *m_write_ep;
    QTimer *timer;

    QByteArray m_send, m_recv;

    QUsbDevice::Id m_filter;
    QUsbDevice::Config m_config;

};
```

fpoussin, 2020-05-14:
With the HID module, you need to specify the read size, or allocate the buffer.
```
    QByteArray array;
    getFeatureReport(&array, 32);
```
or
```
    QByteArray array(32, 0); // 32 bytes filled with 0
    getFeatureReport(&array);
```
Same if you use read()

For QUsbDevice, are you sure the endpoints are correct?

gleke, 2020-05-14:
```
 Device - Device ID (followed by the endpoint for USB devices)
            (12) USB input device
            (15) HID-compliant device
  Address - FireWire async address or channel number
  Length - Total transfer length
  Phase  - Phase Type
            IN    Data in transfer           
  Data   - Hex dump of the data transferred
  Descr  - Description of the phase
  Cmd... - Position in the captured data

Device  Address              Length    Phase  Data  Description       Cmd.Phase.Ofs(rep)
------  -------------------  --------  -----  ----  ----------------  ------------------
  12.2                              1  IN     c2    .                   258361.1.0  
```
For QUsbDevice, I sure the endpoint=2 !

gleke, 2020-05-14:
With the HID module, I specify the read size.
```
 QByteArray array(32, 0); // 32 bytes filled with 0
 getFeatureReport(&array);
```
but can't capture usb device data

fpoussin, 2020-05-14:
Can you show me all debug messages?

gleke, 2020-05-14:
```
void HidExample::getData()
{

    QByteArray array;
    read(&array);
    
    for(int i=0;i<array.size();i++){
        char ch = array[i] ;
        int value = int(ch);
        printf("\n=======getData value:=%d",value);
    }

}
```
all debug messages
=======getData value:=0
=======getData value:=0
=======getData value:=0

fpoussin, 2020-05-14:
There should be all libusb/hidapi and QtUsb log messages as well, not just the example.

gleke, 2020-05-14:
Device - Device ID (followed by the endpoint for USB devices)
            **(12) USB input device
            (15) HID-compliant device**
  Address - FireWire async address or channel number
  Length - Total transfer length
  Phase  - Phase Type
            IN    Data in transfer           
  Data   - Hex dump of the data transferred
  Descr  - Description of the phase
  Cmd... - Position in the captured data

Device  Address              Length    Phase  Data  Description       Cmd.Phase.Ofs(rep)
------  -------------------  --------  -----  ----  ----------------  ------------------
  **12.2                              1  IN     c2    .                   258361.1.0**  

This is a HID device,  the USB input device id is 12,  endpoint=2,I use the QHidDevice class  or  QHIDevice  class????

gleke, 2020-05-14:
you can test by usb mouse.

fpoussin, 2020-05-15:
You won't be able to read/write to a mouse or keyboard on windows, unless you run the app as admin. It's protected from keyloggers.

closed 2020-07-13 by fpoussin