fpoussin/QtUsb #54: Problem to install library on ubuntu

chegewara, 2020-07-12:
Hi, sorry if this is trivial, but i cant solve some problem. 
System:
- ubuntu 20.04
- ubuntu 18.04

when i am trying to install library with `make install` i am getting this error:
```
cd src/ && ( test -e Makefile || /usr/lib/qt5/bin/qmake -o Makefile /home/chegewara/demos/esp32s2/qt_hid/QtUsb/src/src.pro ) && make -f Makefile install
make[1]: Entering directory '/home/chegewara/demos/esp32s2/qt_hid/QtUsb/build/src'
cd usb/ && ( test -e Makefile || /usr/lib/qt5/bin/qmake -o Makefile /home/chegewara/demos/esp32s2/qt_hid/QtUsb/src/usb/usb.pro ) && make -f Makefile install
make[2]: Entering directory '/home/chegewara/demos/esp32s2/qt_hid/QtUsb/build/src/usb'
g++ -c -include .pch/Qt5Usb -g -O2 -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -O2 -std=c++1z -fvisibility=hidden -fvisibility-inlines-hidden -fno-exceptions -Wall -W -Wvla -Wdate-time -Wshift-overflow=2 -Wduplicated-cond -Wno-stringop-overflow -Wno-deprecated-copy -Wno-redundant-move -Wno-format-overflow -Wno-init-list-lifetime -D_REENTRANT -fPIC -DQT_BUILD_USB_LIB -DQT_NO_NARROWING_CONVERSIONS_IN_CONNECT -DQT_BUILD_USB_LIB -DQT_BUILDING_QT -DQT_NO_CAST_TO_ASCII -DQT_ASCII_CAST_WARNINGS -DQT_MOC_COMPAT -DQT_USE_QSTRINGBUILDER -DQT_DEPRECATED_WARNINGS -DQT_DISABLE_DEPRECATED_BEFORE=0x050000 -DQT_NO_EXCEPTIONS -D_LARGEFILE64_SOURCE -D_LARGEFILE_SOURCE -DQT_NO_DEBUG -DQT_CORE_LIB -I/home/chegewara/demos/esp32s2/qt_hid/QtUsb/src/usb -I. -I/home/chegewara/demos/esp32s2/qt_hid/QtUsb/src/usb -I../../include -I../../include/QtUsb -I../../include/QtUsb/0.6.2 -I../../include/QtUsb/0.6.2/QtUsb -isystem /usr/include/libusb-1.0 -isystem /usr/local/include/hidapi -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore/5.12.8 -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore/5.12.8/QtCore -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -I.moc -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o .obj/qusbendpoint.o /home/chegewara/demos/esp32s2/qt_hid/QtUsb/src/usb/qusbendpoint.cpp
In file included from /home/chegewara/demos/esp32s2/qt_hid/QtUsb/src/usb/qusbendpoint.cpp:1:
/home/chegewara/demos/esp32s2/qt_hid/QtUsb/src/usb/qusbendpoint_p.h:6:10: fatal error: private/qiodevice_p.h: No such file or directory
    6 | #include <private/qiodevice_p.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [Makefile:497: .obj/qusbendpoint.o] Error 1
```
im not sure, but most likely its caused by missing `libqt5usb5 libqt5usb5-dev`, which i cant install because of this error:
```
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libqt5usb5 : Depends: hidapi-libusb0 but it is not installable
E: Unable to correct problems, you have held broken packages.
```

Thanks for help

fpoussin, 2020-07-13:
I cannot replicate this on 18.04
This is unrelated to system packages (which you should remove if you install manually)

chegewara, 2020-07-13:
ubuntu 20.04 is my work system, so I installed ubuntu 18.04 on VM, fresh installation, to confirm it is not issue with 20.04 or packages i could have installed.
I believe its problem on my end, since it is working to all others, just cant find on internet solution why im getting this:
```
The following packages have unmet dependencies:
 libqt5usb5 : Depends: hidapi-libusb0 but it is not installable
E: Unable to correct problems, you have held broken packages.
```

Anyway, thanks for help.

chegewara, 2020-07-13:
Maybe its the problem with hidapi library and the way i am installing it?
https://github.com/libusb/hidapi/tree/51bdec78448c7c5e5c16d41e56452befee19ce7f

I follow instructions from it, but maybe there is something else?

fpoussin, 2020-07-13:
I've fixed the package, try again

chegewara, 2020-07-13:
Thanks, now i can install dependencies, but still getting this error:
```
/home/chegewara/demos/esp32s2/qt_hid/QtUsb/src/usb/qusbendpoint_p.h:6:10: fatal error: private/qiodevice_p.h: No such file or directory
    6 | #include <private/qiodevice_p.h>
```

chegewara, 2020-07-13:
Now i can build examples, so its probably fine. Thanks

closed 2020-07-13 by chegewara