fpoussin/QtUsb #58: Closing and reopening device connection causes a deadlock

digitalaspirin, 2020-08-13:
Hi,
It seems like closing and reopening the device causes a deadlock. 

```
// Methods refering to "SimpleBulkExample". 

/* Say you want to close the device due to an error*/
closeDevice (); // returns success

/* Try to re-open device*/
openDevice (); // returns success

/* Write some data to the device.  Eventually, QUsbEndpointPrivate::writeUsb will be called which locks
m_buf_mutex, but cb_out is never executed to release it.*/
m_write_ep->write (...);

/* By calling again write again, it causes a deadlock */
m_write_ep->write (...);
```

digitalaspirin, 2020-08-14:
Hi,
I investigated the reported issue and it seems when the device is closed causes QUsbEventsThread to end execution because QUsbEventsThread::run() exits from while(!this->isInterruptionRequested()) {...} loop, due to ibusb_event_handling_ok(m_ctx) returning '0'. As a result USB events are not processed any more. Hence cb_out is never called.

A quick and dirty fix is to comment out break.

```
void QUsbEventsThread::run()
{
	timeval t = { 0, 100000 };
	while (!this->isInterruptionRequested()) {
		if (libusb_event_handling_ok(m_ctx) == 0) {
			libusb_unlock_events(m_ctx);
			//break; // comment out break; to allow thread to run
		}
		if (libusb_handle_events_timeout_completed(m_ctx, &t, Q_NULLPTR) != 0) {
			break;
		}
	}
}
```
I've also tried the following which it also worked

```
void QUsbDevice::close()
{
	DbgPrintFuncName();
	Q_D(QUsbDevice);

	if (d->m_devHandle && m_connected) {
		// stop any further write attempts whilst we close down
		if (m_log_level >= logInfo)
			qInfo("Closing USB connection");

		libusb_release_interface(d->m_devHandle, 0); // release the claimed interface
		libusb_close(d->m_devHandle); // close the device we opened
+		d->m_events->exit(0);
+		d->m_events->wait();
		d->m_devHandle = Q_NULLPTR;
		m_connected = false;
		emit connectionChanged(m_connected);

	} else {
		if (m_log_level >= logInfo)
			qInfo("USB connection already closed");
	}
}

qint32 QUsbDevice::open()
{
    ...
+        if (!d->m_events->isRunning())
+            d->m_events->start();

	m_connected = true;
	emit connectionChanged(m_connected);

	return 0;

}
```

fpoussin, 2020-08-14:
Thanks for finding this, would you mind opening a pull request with the fix?

fpoussin, 2020-08-17:
Fixed in #59

closed 2020-08-17 by fpoussin