fpoussin/QtUsb #67: Broken MSVC Compilation

gilgm12, 2020-11-24:
**Describe the bug**
`NMAKE : fatal error U1073: don't know how to make 'D:/Projects/Libraries/QtUsb/src/usb/../../libusb/libusb/os/poll_windows.c'`

`10:01:29: Running steps for project QtUsb...
10:01:29: Configuration unchanged, skipping qmake step.
10:01:29: Starting: "E:\Qt\Tools\QtCreator\bin\jom.exe" 
	cd src\ && ( if not exist Makefile E:\Qt\5.14.2\msvc2017_64\bin\qmake.exe -o Makefile D:\Projects\Libraries\QtUsb\src\src.pro -spec win32-msvc "CONFIG+=qtquickcompiler" ) && E:\Qt\Tools\QtCreator\bin\jom.exe -f Makefile
	cd usb\ && ( if not exist Makefile E:\Qt\5.14.2\msvc2017_64\bin\qmake.exe -o Makefile D:\Projects\Libraries\QtUsb\src\usb\usb.pro -spec win32-msvc "CONFIG+=qtquickcompiler" ) && E:\Qt\Tools\QtCreator\bin\jom.exe -f Makefile
	E:\Qt\Tools\QtCreator\bin\jom.exe -f Makefile.Release all
Error: dependent 'D:\Projects\Libraries\QtUsb\src\usb\..\..\libusb\libusb\os\poll_windows.c' does not exist.
jom: D:\Projects\Libraries\build-QtUsb-Desktop_Qt_5_14_2_MSVC2017_64bit-Release\src\usb\Makefile [release-all] Error 2
	E:\Qt\Tools\QtCreator\bin\jom.exe -f Makefile.Debug all
Error: dependent 'D:\Projects\Libraries\QtUsb\src\usb\..\..\libusb\libusb\os\poll_windows.c' does not exist.
jom: D:\Projects\Libraries\build-QtUsb-Desktop_Qt_5_14_2_MSVC2017_64bit-Release\src\usb\Makefile [debug-all] Error 2
jom: D:\Projects\Libraries\build-QtUsb-Desktop_Qt_5_14_2_MSVC2017_64bit-Release\src\Makefile [sub-usb-make_first] Error 2
jom: D:\Projects\Libraries\build-QtUsb-Desktop_Qt_5_14_2_MSVC2017_64bit-Release\Makefile [sub-src-make_first] Error 2
10:01:29: The process "E:\Qt\Tools\QtCreator\bin\jom.exe" exited with code 2.
Error while building/deploying project QtUsb (kit: Desktop Qt 5.14.2 MSVC2017 64bit)
When executing step "Make"
10:01:29: Elapsed time: 00:00.`

**PLatform:**
 - OS: Windows
 - Version Qt 5.14.2 - W10 19042 - vs2017

**To Reproduce**
Steps to reproduce the behavior:
1. Attempt to build in QtCreator or via script.

**Notes**
Latest windows release binaries only contain docs.

gilgm12, 2020-11-24:
Worth noting that libusb@4b64ecc doesn't contain poll_windows.c

fpoussin, 2020-11-24:
Thank, having a look

fpoussin, 2020-11-24:
Fixed

closed 2020-11-24 by fpoussin