fpoussin/QtUsb #68: Avoid keyword 'interface' in QUsb::Config

gilgm12, 2020-11-24:
Triggers compiler error C2236.

The offending classmember is here:
qusb.h:39

When including windows headers, the identifier name "interface" breaks the build.  Swapping the classmember to 'usb_interface' to correct this issue, although I'm not sure that identifier is best or what that particular data member represents. (usb IF was an assumption)

`
        quint8 usb_interface;
`

**PLatform:**
 - OS: Windows
 - Version 10 19042

gilgm12, 2020-11-24:
Optional Patch:
[usb_interface.txt](https://github.com/fpoussin/QtUsb/files/5593106/usb_interface.txt)

hua52841422, 2020-11-25:
INCLUDEPATH += $$PWD/../deps/msvc $$LIBUSB_ROOT_REL $$LIBUSB_ROOT_REL/libusb $$HIDAPI_ROOT_REL $$HIDAPI_ROOT_REL/hidapi  MSVC编译出错

fpoussin, 2020-11-27:
I've had this problem a while ago, I'll need to undefine "interface"

fpoussin, 2020-11-27:
Can you try again?

gilgm12, 2020-11-29:
Yessir - I'll give it a go.  I'm in the middle of a few other adjustments also - is there a good way to connect with you via chat?

fpoussin, 2020-12-05:
Not really, here is fine.
Did you have a chance to retry?

closed 2020-12-17 by fpoussin