fpoussin/QtUsb #69: events_windows.c and windows_common.c missing

neomissing, 2020-12-16:
events_windows.c and windows_common is not exist,Failed to compile

fpoussin, 2020-12-16:
Works fine in CI. You a probably doing something wrong.

neomissing, 2020-12-16:
I am download the code from the git directly, directly in the source code to compile and file the lack of reason to find, source specified libusb data source is not your own libusb data sources, so there is libusb content is not unified, but I changed your own libusb data source, find or cannot be compiled, is directly with hid the demo and I use compile the source code, a pile of warning, not compile in the past, the early version is also not the problem, can also be modified to support mingw compilation, new update with qt call MSVC compilers are not?I want to see where it doesn't match

neomissing@163.com
 
From: Fabien Poussin
Date: 2020-12-16 20:26
To: fpoussin/QtUsb
CC: neomissing; Author
Subject: Re: [fpoussin/QtUsb] events_windows.c and windows_common.c missing (#69)
Works fine in CI. You a probably doing something wrong.

fpoussin, 2020-12-16:
Did you try using the msvc build script?

neomissing, 2020-12-16:
[QtUsb_Hiddemo.zip](https://github.com/fpoussin/QtUsb/files/5702916/QtUsb_Hiddemo.zip)
Attached is my test demo using QT MSVC2019

fpoussin, 2020-12-16:
I cannot check your file now as I'm on my phone. Make sure you initialize the git submodules and use the build script without any modifications. It should work fine. You can also use one of the releases. I can't help you if you change the libusb sources.

neomissing, 2020-12-16:
libusb for libusb source is different with your libusb source，you can compare，events_windows.c is remove，windows_common.c change to windows_nt_common.c，your own libusb source not the same

发自我的iPhone

fpoussin, 2020-12-16:
Yes, this is intended and normal.

neomissing, 2020-12-16:
> Yes, this is intended and normal.

so，the demo is not change anything，but can bulid，if you have time，can test it，think you

neomissing, 2020-12-17:
I use the latest program, recompile, still cannot compile the compiler in the past, a lot of errors, can you give me a normal demo of the compilation，
[image]
[image]
[image]
[image]
[image]

neomissing, 2020-12-17:
I have seen git in CI work is wrong, and the wrong phenomenon is the same as mine. You can check what is wrong, thank you

fpoussin, 2020-12-17:
Yes, I updated the libusb submodule and will update project files. Again, this is normal.

fpoussin, 2020-12-17:
Fixed

closed 2020-12-17 by fpoussin

neomissing, 2020-12-18:
Hello, there are still two bug points modified, such as change,(1)qusb.h line 6 add #include <QObject>,(2)qusb.cpp line 4 add #include "qdebug.h" after modification to support qt MSVC compilation

neomissing@163.com
 
From: Fabien Poussin
Date: 2020-12-17 23:01
To: fpoussin/QtUsb
CC: neomissing; Author
Subject: Re: [fpoussin/QtUsb] events_windows.c and windows_common.c missing (#69)
Fixed

neomissing, 2020-12-18:
There is also a problem with the signals being sent repeatedly. For example, in the UsbNotifications demo, plug and pull USB will trigger two plug and two pull signals

neomissing@163.com
 
From: neomissing@163.com
Date: 2020-12-18 12:24
To: fpoussin/QtUsb
Subject: Re: Re: [fpoussin/QtUsb] events_windows.c and windows_common.c missing (#69)

Hello, there are still two bug points modified, such as change,(1)qusb.h line 6 add #include <QObject>,(2)qusb.cpp line 4 add #include "qdebug.h" after modification to support qt MSVC compilation

neomissing@163.com
 
From: Fabien Poussin
Date: 2020-12-17 23:01
To: fpoussin/QtUsb
CC: neomissing; Author
Subject: Re: [fpoussin/QtUsb] events_windows.c and windows_common.c missing (#69)
Fixed