fpoussin/QtUsb #71: It seems not working on debian buster

Hemingway2003, 2021-01-26:
When I git clone the code and run:
```
mkdir build
cd build
qmake ..
make
```
The system told me `fatal error: private/qobject_p.h: No such file or directory`
And I re-edit the makefile under `build/src/usb/Makefile`, add the `-I` to include these path
But the system showed me more errors

Hemingway2003, 2021-01-26:
Here is the errors:
```
$ make
cd src/ && ( test -e Makefile || /usr/lib/qt5/bin/qmake -o Makefile /home/hemingway/github/QtUsb/src/src.pro ) && make -f Makefile 
make[1]: Entering directory '/home/hemingway/github/QtUsb/build/src'
cd usb/ && ( test -e Makefile || /usr/lib/qt5/bin/qmake -o Makefile /home/hemingway/github/QtUsb/src/usb/usb.pro ) && make -f Makefile 
make[2]: Entering directory '/home/hemingway/github/QtUsb/build/src/usb'
perl /usr/lib/x86_64-linux-gnu/qt5/mkspecs/features/data/unix/findclasslist.pl < QtUsb.version.in > QtUsb.version
g++ -g -O2 -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -O2 -std=c++1z -fvisibility=hidden -fvisibility-inlines-hidden -fno-exceptions -Wall -W -Wvla -Wdate-time -Wshift-overflow=2 -Wduplicated-cond -Wno-stringop-overflow -D_REENTRANT -fPIC -DQT_BUILD_USB_LIB -DQT_NO_NARROWING_CONVERSIONS_IN_CONNECT -DQT_BUILD_USB_LIB -DQT_BUILDING_QT -DQT_NO_CAST_TO_ASCII -DQT_ASCII_CAST_WARNINGS -DQT_MOC_COMPAT -DQT_USE_QSTRINGBUILDER -DQT_DEPRECATED_WARNINGS -DQT_DISABLE_DEPRECATED_BEFORE=0x050000 -DQT_NO_EXCEPTIONS -D_LARGEFILE64_SOURCE -D_LARGEFILE_SOURCE -DQT_NO_DEBUG -DQT_CORE_LIB -I /home/hemingway/software/Qt5.14.2/5.14.2/gcc_64/include/QtCore/5.14.2 -I /home/hemingway/software/Qt5.14.2/5.14.2/gcc_64/include/QtCore/5.14.2/QtCore -I/home/hemingway/github/QtUsb/src/usb -I. -I/home/hemingway/github/QtUsb/src/usb -I../../include -I../../include/QtUsb -I../../include/QtUsb/0.7.0 -I../../include/QtUsb/0.7.0/QtUsb -isystem /usr/include/libusb-1.0 -isystem /usr/include/hidapi -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore/5.11.3 -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore/5.11.3/QtCore -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -I.moc -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -x c++-header -c ../../include/QtUsb/QtUsbDepends -o .pch/Qt5Usb.gch/c++
g++ -c -include .pch/Qt5Usb -g -O2 -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -O2 -std=c++1z -fvisibility=hidden -fvisibility-inlines-hidden -fno-exceptions -Wall -W -Wvla -Wdate-time -Wshift-overflow=2 -Wduplicated-cond -Wno-stringop-overflow -D_REENTRANT -fPIC -DQT_BUILD_USB_LIB -DQT_NO_NARROWING_CONVERSIONS_IN_CONNECT -DQT_BUILD_USB_LIB -DQT_BUILDING_QT -DQT_NO_CAST_TO_ASCII -DQT_ASCII_CAST_WARNINGS -DQT_MOC_COMPAT -DQT_USE_QSTRINGBUILDER -DQT_DEPRECATED_WARNINGS -DQT_DISABLE_DEPRECATED_BEFORE=0x050000 -DQT_NO_EXCEPTIONS -D_LARGEFILE64_SOURCE -D_LARGEFILE_SOURCE -DQT_NO_DEBUG -DQT_CORE_LIB -I /home/hemingway/software/Qt5.14.2/5.14.2/gcc_64/include/QtCore/5.14.2 -I /home/hemingway/software/Qt5.14.2/5.14.2/gcc_64/include/QtCore/5.14.2/QtCore -I/home/hemingway/github/QtUsb/src/usb -I. -I/home/hemingway/github/QtUsb/src/usb -I../../include -I../../include/QtUsb -I../../include/QtUsb/0.7.0 -I../../include/QtUsb/0.7.0/QtUsb -isystem /usr/include/libusb-1.0 -isystem /usr/include/hidapi -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore/5.11.3 -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore/5.11.3/QtCore -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -I.moc -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o .obj/qusb.o /home/hemingway/github/QtUsb/src/usb/qusb.cpp
In file included from /home/hemingway/software/Qt5.14.2/5.14.2/gcc_64/include/QtCore/5.14.2/QtCore/private/qobject_p.h:55,
                 from /home/hemingway/github/QtUsb/src/usb/qusb_p.h:5,
                 from /home/hemingway/github/QtUsb/src/usb/qusb.cpp:2:
/home/hemingway/software/Qt5.14.2/5.14.2/gcc_64/include/QtCore/5.14.2/QtCore/private/qglobal_p.h:77:19: error: missing binary operator before token "("
 #if !__has_builtin(__builtin_available)
                   ^
In file included from /usr/include/x86_64-linux-gnu/qt5/QtCore/QtCore:4,
                 from ../../include/QtUsb/QtUsbDepends:3:
/home/hemingway/software/Qt5.14.2/5.14.2/gcc_64/include/QtCore/5.14.2/QtCore/private/qobject_p.h:490:5: error: division by zero in #if
 #if QT_CONFIG(thread)
     ^~~~~~~~~
/home/hemingway/software/Qt5.14.2/5.14.2/gcc_64/include/QtCore/5.14.2/QtCore/private/qobject_p.h:504:5: error: division by zero in #if
 #if QT_CONFIG(thread)
     ^~~~~~~~~
In file included from /usr/include/x86_64-linux-gnu/qt5/QtCore/QtCore:4,
                 from ../../include/QtUsb/QtUsbDepends:3:
/home/hemingway/software/Qt5.14.2/5.14.2/gcc_64/include/QtCore/5.14.2/QtCore/private/qobject_p.h: In member function ‘void QObjectPrivate::Connection::deref()’:
/home/hemingway/software/Qt5.14.2/5.14.2/gcc_64/include/QtCore/5.14.2/QtCore/private/qobject_p.h:187:36: error: ‘class QAtomicPointer<QObject>’ has no member named ‘loadRelaxed’
                 Q_ASSERT(!receiver.loadRelaxed());
                                    ^~~~~~~~~~~
In file included from /home/hemingway/github/QtUsb/src/usb/qusb_p.h:5,
                 from /home/hemingway/github/QtUsb/src/usb/qusb.cpp:2:
/home/hemingway/software/Qt5.14.2/5.14.2/gcc_64/include/QtCore/5.14.2/QtCore/private/qobject_p.h: In constructor ‘QObjectPrivate::Sender::Sender(QObject*, QObject*, int)’:
/home/hemingway/software/Qt5.14.2/5.14.2/gcc_64/include/QtCore/5.14.2/QtCore/private/qobject_p.h:205:70: error: ‘class QAtomicPointer<QObjectPrivate::ConnectionData>’ has no member named ‘loadRelaxed’
                 ConnectionData *cd = receiver->d_func()->connections.loadRelaxed();
[... cut ...]
/home/hemingway/software/Qt5.14.2/5.14.2/gcc_64/include/QtCore/5.14.2/QtCore/private/qobject_p.h:557:37: error: expected ‘;’ at end of member declaration
     Q_DISABLE_COPY_MOVE(QBoolBlocker)
                                     ^
                                      ;
make[2]: *** [Makefile:503: .obj/qusb.o] Error 1
make[2]: Leaving directory '/home/hemingway/github/QtUsb/build/src/usb'
make[1]: *** [Makefile:45: sub-usb-make_first] Error 2
make[1]: Leaving directory '/home/hemingway/github/QtUsb/build/src'
make: *** [Makefile:47: sub-src-make_first] Error 2

```

fpoussin, 2021-01-31:
Hi,
It just seems like you are missing the private headers package

closed 2021-02-04 by fpoussin