fpoussin/QtUsb #72: Qt Creator build failure

silverdr, 2021-02-07:
**Describe the bug**
When trying to build `QtUsb` I receive an error about not being able to find `libusb-1.0`
```
21:57:17: Running steps for project QtUsb...
21:57:17: Configuration unchanged, skipping qmake step.
21:57:17: Starting: "/usr/bin/make" -j24
cd src/ && ( test -e Makefile || /Users/silverdr/sources/Qt5.9.9/5.9.9/clang_64/bin/qmake -o Makefile /Users/silverdr/sources/QtUsb/src/src.pro -spec macx-clang CONFIG+=debug CONFIG+=x86_64 CONFIG+=qml_debug ) && /Volumes/Scratch/Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile 
cd usb/ && ( test -e Makefile || /Users/silverdr/sources/Qt5.9.9/5.9.9/clang_64/bin/qmake -o Makefile /Users/silverdr/sources/QtUsb/src/usb/usb.pro -spec macx-clang CONFIG+=debug CONFIG+=x86_64 CONFIG+=qml_debug ) && /Volumes/Scratch/Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile 
Project MESSAGE: Build as Qt module for macos osx macx mac darwin unix posix
Project ERROR: Could not find libusb-1.0 using pkg-config
make[1]: *** [sub-usb-make_first] Error 3
make: *** [sub-src-make_first] Error 2
21:57:18: The process "/usr/bin/make" exited with code 2.
Error while building/deploying project QtUsb (kit: Desktop Qt 5.9.9 clang 64bit)
When executing step "Make"
21:57:18: Elapsed time: 00:02.
```
Of course I have `libusb-1.0` installed and `pkg-config` finds it easily when invoked from the command line:

```
silverdr$ pkg-config --cflags libusb-1.0
-I/usr/local/Cellar/libusb/1.0.24/include/libusb-1.0
```

**PLatform:**
 - OS: OS X
 - Version: 10.14.6

**To Reproduce**
Steps to reproduce the behavior:
1. clone the repository afresh
2. open QtUsb project in Qt Creator
3. Invoke Build
4. See error

**Expected behavior**
Project with all dependencies gets built

**Additional context**
I am new to Qt and Qt Creator so this might be a factor that I am missing something "obvious"

fpoussin, 2021-02-08:
Can you try with a more recent Qt version?
Looks like a qmake bug.

silverdr, 2021-02-09:
Tnx. It is 5.9.x so not totally ancient but I'll try on 5.15 (I guess is the current 5.x one?) and report. It will take some time so please don't close too early :-)

fpoussin, 2021-02-12:
Give this a try :) 
https://stackoverflow.com/questions/41593607/qt-creator-on-macos-pkg-config-dependencies-cant-be-found

silverdr, 2021-02-17:
I gave it a try. Added manually `/usr/local/bin`. pkg-config seem to get through now but I still am unable to build due to:
```
[…]/QtUsb/src/usb/qusbdevice_p.h:9: error: 'libusb-1.0/libusb.h' file not found
In file included from /Users/silverdr/sources/QtUsb/src/usb/qusbdevice.cpp:2:
[…]/QtUsb/src/usb/qusbdevice_p.h:9:10: fatal error: 'libusb-1.0/libusb.h' file not found
#include <libusb-1.0/libusb.h>
         ^~~~~~~~~~~~~~~~~~~~~
```
Shall check more thoroughly later.

fpoussin, 2021-03-24:
I've fixed the imports for MacOS. Builds fine now (with the pkg-config path added to the project)

closed 2021-03-24 by fpoussin