fpoussin/QtUsb #79: Maybe you need to provide an instruction to build QtUSB by using pre-built version of hidapi and libusb.

jerry-yuan, 2021-06-14:
I am using mingw to compile the QtUSB, while the project choose to build hidapi and libusb itself. There are lots of errors when I try to build the libusb, but I have installed a version of libusb. I have changed the following codes to make it could be built in mingw32. I wish you can add some settings to make it can be changed to build with existed librarys.
```diff
diff --git a/src/usb/files.pri b/src/usb/files.pri
index 23aae8b..ff38e3e 100644
--- a/src/usb/files.pri
+++ b/src/usb/files.pri
@@ -53,26 +53,27 @@ SOURCES += \
     $$PWD/qhiddevice.cpp
 
 win32 {
-    LIBS_PRIVATE += Advapi32.lib Setupapi.lib
+    LIBS_PRIVATE += -ladvapi32 -lsetupapi -lhidapi -lusb-1.0
+    
     # Build the hid library ourselves since it's only one file
-    HIDAPI_ROOT_REL = $$PWD/../../hidapi
-    SOURCES += $$HIDAPI_ROOT_REL/windows/hid.c
-
-    LIBUSB_ROOT_REL = $$PWD/../../libusb
-    SOURCES += \
-    $$LIBUSB_ROOT_REL/libusb/core.c \
-    $$LIBUSB_ROOT_REL/libusb/descriptor.c \
-    $$LIBUSB_ROOT_REL/libusb/os/events_windows.c \
-    $$LIBUSB_ROOT_REL/libusb/hotplug.c \
-    $$LIBUSB_ROOT_REL/libusb/io.c \
-    $$LIBUSB_ROOT_REL/libusb/strerror.c \
-    $$LIBUSB_ROOT_REL/libusb/sync.c \
-    $$LIBUSB_ROOT_REL/libusb/os/threads_windows.c \
-    $$LIBUSB_ROOT_REL/libusb/os/windows_common.c \
-    $$LIBUSB_ROOT_REL/libusb/os/windows_usbdk.c \
-    $$LIBUSB_ROOT_REL/libusb/os/windows_winusb.c
-
-    INCLUDEPATH += $$PWD/../deps/msvc $$LIBUSB_ROOT_REL $$LIBUSB_ROOT_REL/libusb $$HIDAPI_ROOT_REL $$HIDAPI_ROOT_REL/hidapi
+    #HIDAPI_ROOT_REL = $$PWD/../../hidapi
+    #SOURCES += $$HIDAPI_ROOT_REL/windows/hid.c
+
+    #LIBUSB_ROOT_REL = $$PWD/../../libusb
+    #SOURCES += \
+    #$$LIBUSB_ROOT_REL/libusb/core.c \
+    #$$LIBUSB_ROOT_REL/libusb/descriptor.c \
+    #$$LIBUSB_ROOT_REL/libusb/os/events_windows.c \
+    #$$LIBUSB_ROOT_REL/libusb/hotplug.c \
+    #$$LIBUSB_ROOT_REL/libusb/io.c \
+    #$$LIBUSB_ROOT_REL/libusb/strerror.c \
+    #$$LIBUSB_ROOT_REL/libusb/sync.c \
+    #$$LIBUSB_ROOT_REL/libusb/os/threads_windows.c \
+    #$$LIBUSB_ROOT_REL/libusb/os/windows_common.c \
+    #$$LIBUSB_ROOT_REL/libusb/os/windows_usbdk.c \
+    #$$LIBUSB_ROOT_REL/libusb/os/windows_winusb.c
+
+    #INCLUDEPATH += $$PWD/../deps/msvc $$LIBUSB_ROOT_REL $$LIBUSB_ROOT_REL/libusb $$HIDAPI_ROOT_REL $$HIDAPI_ROOT_REL/hidapi
 }
 
 # We build libusb and hidapi ourselves instead of using a library
diff --git a/src/usb/qhiddevice_p.h b/src/usb/qhiddevice_p.h
index d24996f..b86b748 100644
--- a/src/usb/qhiddevice_p.h
+++ b/src/usb/qhiddevice_p.h
@@ -3,7 +3,7 @@
 
 #include "qhiddevice.h"
 #include <private/qobject_p.h>
-#include <hidapi.h>
+#include <hidapi/hidapi.h>
 
 QT_BEGIN_NAMESPACE
 
diff --git a/src/usb/qusb.cpp b/src/usb/qusb.cpp
index c3ce376..3df40a8 100644
--- a/src/usb/qusb.cpp
+++ b/src/usb/qusb.cpp
@@ -1,6 +1,7 @@
 #include "qusb.h"
 #include "qusb_p.h"
 #include <QThread>
+#include <QDebug>
 
 #define DbgPrintError() qWarning("In %s, at %s:%d", Q_FUNC_INFO, __FILE__, __LINE__)
 #define DbgPrintFuncName()                 \
diff --git a/src/usb/qusb.h b/src/usb/qusb.h
index 8175364..3e63977 100644
--- a/src/usb/qusb.h
+++ b/src/usb/qusb.h
@@ -3,6 +3,7 @@
 
 #include "qusbglobal.h"
 #include <QList>
+#include <QObject>
 
 // Stupid windows conflict
 #ifdef interface
diff --git a/src/usb/qusb_p.h b/src/usb/qusb_p.h
index b82996c..ad604b1 100644
--- a/src/usb/qusb_p.h
+++ b/src/usb/qusb_p.h
@@ -12,7 +12,7 @@
   #include <libusb-1.0/libusb.h>
   #include <hidapi.h>
 #else
-  #include <libusb/libusb.h>
+  #include <libusb-1.0/libusb.h>
   #include <hidapi/hidapi.h>
 #endif
 
diff --git a/src/usb/qusbdevice_p.h b/src/usb/qusbdevice_p.h
index 3bf4bb2..d4bfcee 100644
--- a/src/usb/qusbdevice_p.h
+++ b/src/usb/qusbdevice_p.h
@@ -4,13 +4,14 @@
 #include "qusbdevice.h"
 #include <private/qobject_p.h>
 #include <QThread>
+#include <QObject>
 
 #if defined(Q_OS_MACOS)
   #include <libusb.h>
 #elif defined(Q_OS_UNIX)
   #include <libusb-1.0/libusb.h>
 #else
-  #include <libusb/libusb.h>
+  #include <libusb-1.0/libusb.h>
 #endif
 
 QT_BEGIN_NAMESPACE
diff --git a/src/usb/qusbendpoint_p.h b/src/usb/qusbendpoint_p.h
index 7dd6728..a838da8 100644
--- a/src/usb/qusbendpoint_p.h
+++ b/src/usb/qusbendpoint_p.h
@@ -10,7 +10,7 @@
 #elif defined(Q_OS_UNIX)
   #include <libusb-1.0/libusb.h>
 #else
-  #include <libusb/libusb.h>
+  #include <libusb-1.0/libusb.h>
 #endif
 
 QT_BEGIN_NAMESPACE
```

Besides, I got some errors (I have fixed myself) triggered by the lacking of QObject and QDebug when I build it by mingw. I want to know if there has the same errors when you build with msvc? 

I have checked the examples and it can work well, so I wish you can add the support for mingw32 and the instruction to build it with pre-built version of third-party libs, thank you.

jerry-yuan, 2021-06-14:
I have commited my modification to my fork repo https://github.com/jerry-yuan/QtUsb.

I suprisingly find that you added the hotplug support on windows which was not supported by libusb. THANKS FOR YOUR WORK!!!! With this feature, I can complete my application easily.

fpoussin, 2021-07-10:
Hi,
This is expected, the windows version uses MSVC.
It's possible to specify the compiler by replacing win32 {} by msvc {} and adding a mingw {} section in the .pri file