fpoussin/QtUsb #80: QUSB will block the main thread and lead to stuck of ui.

jerry-yuan, 2021-06-15:
**Describe the bug**
In qusb.usb@261 a QTimer with timeout of 250ms has connected to a slot `checkDevices`, in which will invoke the `libusb_handle_events_timeout_completed` on platforms libusb support hotplug detection (like linux) or manually detect the hotplug by polling the device lists on platrforms libusb that do not support hotplug detect (like windows):
```cpp

/*!
    Check devices present in system.

    This gets called by the internal timer.
 */
void QUsb::checkDevices()
{
    DbgPrintFuncName();
    Q_D(QUsb);
    QUsb::IdList list;

    timeval t = { 0, 0 };
    qDebug()<<QThread::currentThread();
    if (d->m_has_hotplug) {
        libusb_handle_events_timeout_completed(d->m_ctx, &t, Q_NULLPTR);
    } else {
        list = devices();
        monitorDevices(list);
    }
}
```
However, this slot function will run in the main thread where the ui will be rendered:
```
[DEBUG][2021-06-15 17:10:02.338]:main.cpp@11: QThread(0x1e20b2f8)
[DEBUG][2021-06-15 17:10:02.489]:qusb.cpp@286: QThread(0x1e20b2f8)
[DEBUG][2021-06-15 17:10:02.847]:qusb.cpp@286: QThread(0x1e20b2f8)
[DEBUG][2021-06-15 17:10:03.140]:qusb.cpp@286: QThread(0x1e20b2f8)
[DEBUG][2021-06-15 17:10:03.473]:qusb.cpp@286: QThread(0x1e20b2f8)
```
 For this reason, if I instantance a QUsb object there will be a time-consuming process that need to be processed in main thread and the UI rendering process will be disturbed. This problem will lead to an UI stuck.

**PLatform:**
 - OS: Platforms that use polling to detect hotplug, e.g. Windows
 - Version: 10

**To Reproduce**
Steps to reproduce the behavior:
1. Create a Qt Widget Application with default settings.
2. add QtUsb Module to .pro file:
    ```cpp
    QT       += core gui usb
    ```
3. Instance the QUsb in class `MainWindow` or `MainDialog`
    ```cpp
   #ifndef MAINWINDOW_H
#define MAINWINDOW_H

#include <QMainWindow>
#include <QUsb>
QT_BEGIN_NAMESPACE
namespace Ui { class MainWindow; }
QT_END_NAMESPACE

class MainWindow : public QMainWindow
{
    Q_OBJECT

public:
    MainWindow(QWidget *parent = nullptr);
    ~MainWindow();

private:
    Ui::MainWindow *ui;
    QUsb usbInstance;
};
#endif // MAINWINDOW_H
```
3. Compile and Run the program,when the window showed, drag the window you will find the window does not follow the cursor smoothly:
![with QUsb initialized](https://user-images.githubusercontent.com/13380647/122029865-d6db8f80-cdff-11eb-98fc-bc03fcacd917.gif)
While remove the instance of QUsb just run an empty Qt Widget Application the window could be drag smoothly:
![GIF](https://user-images.githubusercontent.com/13380647/122030344-481b4280-ce00-11eb-9763-ded39a32d996.gif)
Further more, you can try to instance more QUsb objects to amplify the jam caused by polling.

4. See error

**Expected behavior**
I'd like this process not running in the main process.

**Additional context**
I have checked out the sources, I found the timer has been moved to another thread while the QUsb has not moved to the other thread.
In the previous test case, if you add a QThread in MainWindow and move QUsb into it would solve the jam problem, but maybe this thread should be mantainanced by QUsb?

fpoussin, 2021-07-09:
Thanks, I'll have a look

Alvazz, 2022-11-11:
@jerry-yuan Hi, Jerry. did you finally solve this problem? How was it resolved

jerry-yuan, 2022-11-11:
> @jerry-yuan Hi, Jerry. did you finally solve this problem? How was it resolved

Yes, I have solved this problem by moving the monitor function to another scanning thread.

closed 2026-03-15 by jerry-yuan