fpoussin/QtUsb #88: how to use the QUsbEndpoint::controlEndpoint

Alvazz, 2022-11-17:
Hi fpoussin. Thank for you shared this Good project.

I want use the QUsbEndpoint::controlEndpoint way. can you give me some demo code?

thanks!

fpoussin, 2023-02-04:
Hi,

You can use the `makeControlPacket` method to generate control packets, then write it just like a normal endpoint.
It mostly depends on the device.

closed 2023-02-04 by fpoussin

Alvazz, 2023-02-27:
[image]

```
m_read_ep = new QUsbEndpoint(m_usb_dev,QUsbEndpoint::controlEndpoint,0);
m_read_ep->makeControlPacket(buffer,QUsbEndpoint::recipientInterface,QUsbEndpoint::requestClearFeature,0x0603, 0x0600, 0x3E00);
m_read_ep->write(buffer);
```

Should I write like this?

Then where do I receive and return the data?