fpoussin/QtUsb #96: Feature: FTDI line/modem status byte dropping

petrmanek, 2023-10-17:
**Is your feature request related to a problem? Please describe.**
Yes, I am trying to communicate with a device that uses a FT232H serial chip and have been encountering excess data in my read() calls. After some poking around, I discovered out that FTDI serial protocol always returns a 2-byte-long header in all read() calls containing flow control flags and other auxiliary information, followed by actual data, in which I am intersted. `QUsbEndpoint`'s read callback incorrectly combines this 2-byte header with actual data and adds them to the read buffer of `QIODevice` without delimiters. This means that while my app is polling for data, the read buffer is slowly filling with information that is not used for anything relevant. Additionally, since the user code does not see where one callback ends and other begins, the headers cannot be easily separated from data.

**Describe the solution you'd like**
I would propose that `QUsbEndpoint` is extended to prevent from receiving FTDI 2-byte headers in the first place, i.e. to pretend that the first 2 bytes of any received payload do not exist. Of course, since not all QtUsb users communicate with FTDI chips, this would have to be an opt-in feature that would be relevant for FTDI hardware only.

Several possible solutions come to mind:

1. Create `QFtdiEndpoint` as a subclass of `QUsbEndpoint` with the only change being that it uses a different callback under the hood. This would also give us future extension points to implement other FTDI-specific features in that class.
2. Extend `QUsbEndpoint` with something like `droppedPrefixSize` property that would be configured to 0 by default, and could be set to 2 for FTDI chips. This would have slightly smaller footprint on the codebase but is not perfect code architecture in my humble opinion.
3. Extend `QUsbEndpoint` with a callback property (or delegate) that allows user code to transform the payload between the point of reception from libusb, and before it gets appended to the read buffer of `QIODevice`. By default this would be an identity mapping that does not do anything, but users like me could override it to strip away the header. This is the most generic solution, which would unfortunately incur the most severe performance penalty (1 function pointer jump per every received payload).

**Describe alternatives you've considered**
See the 3 options outlined above. I have also considered using libftdi directly, however that does not work nicely with an app that already heavily relies on QtUsb. It would be nice to solve this here.

**Additional context**
* [Here is how libftdi handles incoming FTDI data from libusb](https://github.com/particle-iot/libftdi/blob/master/src/ftdi.c#L1616). I would consider this to be the reference implementation of the behavior that I want to achieve.
* [Here is the callback in QtUsb that I needed to patch](https://github.com/fpoussin/QtUsb/blob/master/src/usb/qusbendpoint.cpp#L85)

And here is my patched version (only proof of concept) that only works for FTDI chips (note all the +2's and -2's):

```cpp
    if (s != LIBUSB_TRANSFER_COMPLETED) {
        endpoint->error(static_cast<QUsbEndpoint::Status>(s));
    } else {
        if (received > 2) {
            endpoint->m_buf_mutex.lock();
            const int previous_size = endpoint->m_buf.size();
            endpoint->m_buf.resize(previous_size + received - 2);
            memcpy(endpoint->m_buf.data() + previous_size, transfer->buffer + 2, static_cast<ulong>(received - 2));
            endpoint->m_buf_mutex.unlock();
        }
    }

    libusb_free_transfer(transfer);
    endpoint->m_transfer = Q_NULLPTR;

    endpoint->m_transfer_buf.clear(); // it's in fact transfer->buffer
    endpoint->m_transfer_mutex.unlock();

    if (received > 2)
        endpoint->readyRead();
```

petrmanek, 2023-10-18:
One more relevant link: [Here is how the 2-byte stripping is handled in pyftdi](https://github.com/eblot/pyftdi/blob/master/pyftdi/ftdi.py#L1857).

petrmanek, 2023-10-23:
@fpoussin Bumping this in case it got lost in the inbox.

petrmanek, 2023-11-22:
@fpoussin Anyone here?

petrmanek, 2024-10-21:
@fpoussin It has been a year since my last message. Can I please get _any_ response, whether you would be interested in this?

fpoussin, 2024-10-22:
Hi,
Sorry for the late response, I haven't had much time for this.
As QtUSB is a generic USB library and not dedicated to serial communication, I feel like this would be out of scope.
You'd probably best inherit QUsbEndpoint and extend it or use QSerialDevice with the ftdi driver.

petrmanek, 2024-10-22:
Hello, thanks for your reply. I will close this then!

closed 2024-10-22 by petrmanek